openhab/org.openhab.binding.zwave #552: OH2.1 FGMS-001 motion detecting problem

brutevinch, 2017-04-21:
* OH2 Version: 2.1
* Fibaro FGMS-001
When motion is detected OH2 not update channel alarm_motion only alarm tamper

15:36:50.409 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Initialising cmd channel zwave:device:15b7bb97064:node5:alarm_motion
15:36:50.410 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Initialising poll channel zwave:device:15b7bb97064:node5:alarm_motion
15:36:50.411 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Initialising state channel zwave:device:15b7bb97064:node5:alarm_motion
15:36:50.411 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Initialising state channel zwave:device:15b7bb97064:node5:alarm_motion

15:45:14.673 [DEBUG] [dclass.ZWaveBinarySensorCommandClass] - NODE 5: Received SENSOR_BINARY command V1
15:45:14.674 [DEBUG] [dclass.ZWaveBinarySensorCommandClass] - NODE 5: Sensor Binary report, type=Unknown, value=255
15:45:14.676 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveBinarySensorValueEvent
15:45:14.677 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 255
15:45:14.679 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:15b7bb97064:node5:alarm_tamper to ON [OnOffType]
15:45:44.982 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Application Command Request (ALIVE:DETAILS)
15:45:44.983 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Incoming command class CRC_16_ENCAP
15:45:44.983 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 5: Received CRC 16 Encapsulation Request
15:45:44.984 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 5: Calling handleApplicationCommandRequest.
15:45:44.985 [DEBUG] [dclass.ZWaveBinarySensorCommandClass] - NODE 5: Received SENSOR_BINARY command V1
15:45:44.986 [DEBUG] [dclass.ZWaveBinarySensorCommandClass] - NODE 5: Sensor Binary report, type=Unknown, value=0
15:45:44.987 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveBinarySensorValueEvent
15:45:44.988 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 0
15:45:44.989 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:15b7bb97064:node5:alarm_tamper to OFF [OnOffType]

cdjackson, 2017-04-21:
Please try the development branch - see the security thread on the forum.

brutevinch, 2017-04-21:
How can i install dev branch of binding or i must to try dev branch of openhab2.1?
Can't find thread =( sorry for disturb

brutevinch, 2017-04-21:
Last openhab 2.1 snapshot give me this log 

ngHandler] - NODE 5: Got an event from Z-Wave network: ZWaveBinarySensorValueEvent
17:12:10.750 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 255
17:12:10.759 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:15b7bb97064:node5:alarm_tamper to ON [OnOffType]
17:12:48.641 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Application Command Request (ALIVE:DETAILS)
17:12:48.641 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Incoming command class CRC_16_ENCAP
17:12:48.642 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 5: Received CRC 16 Encapsulation Request
17:12:48.643 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 5: Calling handleApplicationCommandRequest.
17:12:48.644 [DEBUG] [dclass.ZWaveBinarySensorCommandClass] - NODE 5: Received SENSOR_BINARY command V1
17:12:48.644 [DEBUG] [dclass.ZWaveBinarySensorCommandClass] - NODE 5: Sensor Binary report, type=Unknown, value=0
17:12:48.648 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveBinarySensorValueEvent
17:12:48.650 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 0
17:12:48.653 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:15b7bb97064:node5:alarm_tamper to OFF [OnOffType]

cdjackson, 2017-04-21:
Please try the development branch - see below- :
https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653

brutevinch, 2017-04-21:
Ok. dev branch build installed. Now i've got this:

20:11:34.965 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Node not awake!
20:11:38.138 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Application Command Request (ALIVE:REQUEST_NIF)
20:11:38.138 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: Decapsulating COMMAND_CLASS_CRC_16_ENCAP
20:11:38.139 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: COMMAND_CLASS_CRC_16_ENCAP not found
20:11:38.140 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: listening == false, frequentlyListening == false, awake == false
20:11:38.140 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Node not awake!
20:11:42.752 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Application Command Request (ALIVE:REQUEST_NIF)
20:11:42.753 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: Decapsulating COMMAND_CLASS_CRC_16_ENCAP
20:11:42.753 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: COMMAND_CLASS_CRC_16_ENCAP not found
20:11:42.755 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: listening == false, frequentlyListening == false, awake == false
20:11:42.755 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Node not awake!
20:12:47.767 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Application Command Request (ALIVE:REQUEST_NIF)
20:12:47.768 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: Decapsulating COMMAND_CLASS_CRC_16_ENCAP
20:12:47.772 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: COMMAND_CLASS_CRC_16_ENCAP not found
20:12:47.774 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: listening == false, frequentlyListening == false, awake == false
20:12:47.775 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Node not awake!
20:14:16.093 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Application Command Request (ALIVE:REQUEST_NIF)
20:14:16.096 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: Decapsulating COMMAND_CLASS_CRC_16_ENCAP
20:14:16.098 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: COMMAND_CLASS_CRC_16_ENCAP not found
20:14:16.105 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: listening == false, frequentlyListening == false, awake == false
20:14:16.108 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Node not awake!
20:15:32.971 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Application Command Request (ALIVE:REQUEST_NIF)
20:15:32.974 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: Decapsulating COMMAND_CLASS_CRC_16_ENCAP
20:15:32.976 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: COMMAND_CLASS_CRC_16_ENCAP not found
20:15:32.982 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: listening == false, frequentlyListening == false, awake == false
20:15:32.984 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 5: Node not awake!

cdjackson, 2017-04-21:
Please read the instructions associated with this binding - you need to delete all things and add them back again

brutevinch, 2017-04-22:
Done. But now I've got FGMS-001 as Unknown Device
 Fibaro Multisensor FGMS-001 type=0800, id=1001

cdjackson, 2017-04-22:
Version?

brutevinch, 2017-04-22:
binding is last build from security thread you gave me.

cdjackson, 2017-04-22:
Sorry - of the device, not the binding.

brutevinch, 2017-04-22:
How i can check this? On device under cover i see only battery info and "FGMS-001" No info about version. In ozwcp doesn't see version too =(

cdjackson, 2017-04-22:
You had the type/id information - where did you get this? Normally the binding will print all this together with the version.

brutevinch, 2017-04-22:
In ozwcp: 
[image]

brutevinch, 2017-04-22:
here is oh log DEBUG NODE 2

brutevinch, 2017-04-22:
openhab> log:tail | grep "NODE 2"
01:47:53.331 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: MANUFACTURER not set
01:47:56.498 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Adding to controller queue
01:47:56.516 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Added to queue - size 1
01:47:56.545 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 03 00 15 E9 
01:47:56.616 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Adding to controller queue
01:47:56.617 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Added to queue - size 1
01:47:56.636 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Adding to controller queue
01:47:56.638 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Added to queue - size 2
01:47:56.666 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Adding to controller queue
01:47:56.668 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Added to queue - size 3
01:47:56.686 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Adding to controller queue
01:47:56.688 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Added to queue - size 4
01:47:56.714 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Response processed after 137ms
01:47:56.716 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: TID 0: Transaction completed
01:47:56.717 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: notifyTransactionResponse TID:0 DONE
01:47:56.739 [WARN ] [nal.protocol.ZWaveTransactionManager] - NODE 2: Not initialized (ie node unknown), ignoring message.
01:47:56.747 [WARN ] [nal.protocol.ZWaveTransactionManager] - NODE 2: Not initialized (ie node unknown), ignoring message.
01:47:56.754 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 03 00 20 DC 
01:47:56.839 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Response processed after 61ms
01:47:56.840 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: TID 1: Transaction completed
01:47:56.842 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: notifyTransactionResponse TID:1 DONE
01:47:56.884 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 03 00 56 AA 
01:47:56.955 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Adding to controller queue
01:47:56.958 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Added to queue - size 3
01:47:56.978 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Response processed after 80ms
01:47:56.983 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: TID 4: Transaction completed
01:47:56.985 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: notifyTransactionResponse TID:4 DONE
01:47:56.997 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 05 00 06 96 0F 65 
01:47:57.062 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Response processed after 37ms
01:47:57.063 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: TID 3: Transaction completed
01:47:57.064 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: notifyTransactionResponse TID:3 DONE
01:47:57.076 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 03 00 05 F9 
01:47:57.161 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Response processed after 65ms
01:47:57.163 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: TID 5: Transaction completed
01:47:57.165 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: notifyTransactionResponse TID:5 DONE
01:47:57.191 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 03 00 07 FB 
01:47:57.425 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Adding to controller queue
01:47:57.426 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Added to queue - size 1
01:47:57.438 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Response processed after 231ms
01:47:57.448 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: TID 2: Transaction completed
01:47:57.449 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: notifyTransactionResponse TID:2 DONE
01:47:57.467 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 03 00 02 FE 
01:47:57.565 [INFO ] [age.SerialApiGetInitDataMessageClass] - NODE 2: Node found
01:47:57.607 [DEBUG] [ve.internal.protocol.ZWaveController] - NODE 2: Init node thread start
01:47:57.668 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: Response processed after 182ms
01:47:57.670 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: TID 6: Transaction completed
01:47:57.678 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 255: notifyTransactionResponse TID:6 DONE
01:47:58.497 [DEBUG] [l.initialization.ZWaveNodeSerializer] - NODE 2: Serializing from file /opt/openhab2/userdata/zwave/network_fa7f8c4f__node_2.xml
01:47:58.506 [DEBUG] [l.initialization.ZWaveNodeSerializer] - NODE 2: Error serializing from file: file does not exist.
01:47:58.676 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 2: Starting initialisation from EMPTYNODE
01:47:58.687 [DEBUG] [ve.internal.protocol.ZWaveController] - NODE 2: Init node thread finished
01:47:58.702 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 2: Node advancer - advancing to IDENTIFY_NODE
[... cut ...]
01:49:51.441 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
01:49:51.441 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_BASIC, value = 0
01:49:51.442 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Commands processed 1.
01:49:51.442 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@4ca3c2.

cdjackson, 2017-04-22:
What do the attributes show for this device - the information I want should be easily available in HABmin.

brutevinch, 2017-04-22:
Manufacturer
Type / ID
Firmware Version0.0
Basic ClassBASIC_TYPE_ROUTING_SLAVE
Generic ClassGENERIC_TYPE_SENSOR_BINARY
Specific ClassSPECIFIC_TYPE_ROUTING_SENSOR_BINARY
Using Security
Routing
Listening
Frequently Listening
Beaming
Neighbours 1 nodes  1

brutevinch, 2017-04-22:
[image]

cdjackson, 2017-04-22:
As it’s a battery node, make sure it’s woken up enough so that it gets all the data.

brutevinch, 2017-04-23:
Thx. all works after 31 try of initialization =).

closed 2017-04-23 by brutevinch