openhab/org.openhab.binding.zwave #553: Zooz ZEN07 not reporting consumption

rrgeorge, 2017-04-21:
The Zooz ZEN07 on/off module is supposed to report realtime consumption (Amps,Watts,Voltage) in addition to the cumulative meter (kWh). The meter (kWh) appears to work correctly, but the consumption values (Amps,Watts,Voltage) are always 0. I have included the Node XML as well as a DEBUG log for the device.

* OH2 Version: 2.1.0-SNAPSHOT Build #887
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 2.1.0.201704180815

* Device name: ZEN07 Zooz Z-Wave Plus Mini Plug On / Off Module ZEN07
* Device version: 3.81
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/495
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

[openhab.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/947660/openhab.log.txt)

[node10.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/947643/node10.xml.txt)

cdjackson, 2017-04-21:
This looks like a problem with the device to me.

closed 2017-04-21 by cdjackson

rrgeorge, 2017-04-21:
I see that now. My apologies. I will contact the manufacturer.

cdjackson, 2017-04-21:
No probs.