openhab/org.openhab.binding.zwave #557: GE/Jasco 45613 Dimmer Switch Not in Database

theweez928, 2017-04-23:
I am trying to add a GE/Jasco dimmer switch, there seem to be two model numbers on it one is 45613 and the other is 14321. It appears as an unknown device but generates an XML file. Here are the properties displayed on the Paper UI:

zwave_class_basic	ROUTING_SLAVE
zwave_class_generic	MULTILEVEL_SWITCH
zwave_frequent	false
zwave_neighbours	1,8,9
zwave_version	5.23
zwave_listening	true
zwave_plus_devicetype	LIGHT_DIMMER_SWITCH
zwave_deviceid	12597
zwave_nodeid	11
zwave_routing	true
zwave_beaming	true
zwave_class_specific	POWER_SWITCH_MULTILEVEL
zwave_manufacturer	99
zwave_devicetype	18756

Here is the XML file that was generated for the node:
<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>MULTILEVEL_SWITCH</genericDeviceClass>
    <specificDeviceClass>POWER_SWITCH_MULTILEVEL</specificDeviceClass>
  </deviceClass>
  <homeId>0xccdf8152</homeId>
  <nodeId>11</nodeId>
  <version>4</version>
  <manufacturer>0x63</manufacturer>
  <deviceId>0x3135</deviceId>
  <deviceType>0x4944</deviceType>
  <listening>true</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>true</beaming>
  <maxBaudRate>40000</maxBaudRate>
  <nodeInformationFrame>
    <commandClass>ZWAVE_PLUS_INFO</commandClass>
    <commandClass>CRC_16_ENCAP</commandClass>
    <commandClass>VERSION</commandClass>
    <commandClass>MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>DEVICE_RESET_LOCALLY</commandClass>
    <commandClass>ASSOCIATION</commandClass>
    <commandClass>ASSOCIATION_GROUP_INFO</commandClass>
    <commandClass>POWERLEVEL</commandClass>
    <commandClass>SWITCH_MULTILEVEL</commandClass>
    <commandClass>SWITCH_ALL</commandClass>
    <commandClass>CONFIGURATION</commandClass>
    <commandClass>SCENE_ACTUATOR_CONF</commandClass>
    <commandClass>SCENE_ACTIVATION</commandClass>
    <commandClass>FIRMWARE_UPDATE_MD</commandClass>
  </nodeInformationFrame>
  <supportedCommandClasses>
    <entry>
      <commandClass>VERSION</commandClass>
      <versionCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <libraryType>LIB_SLAVE_ENHANCED</libraryType>
        <protocolVersion>4.34</protocolVersion>
        <applicationVersion>5.23</applicationVersion>
        <hardwareVersion>255</hardwareVersion>
      </versionCommandClass>
    </entry>
    <entry>
      <commandClass>BASIC</commandClass>
      <basicCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </basicCommandClass>
    </entry>
    <entry>
      <commandClass>DEVICE_RESET_LOCALLY</commandClass>
      <deviceResetLocallyCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </deviceResetLocallyCommandClass>
    </entry>
    <entry>
      <commandClass>ASSOCIATION</commandClass>
      <associationCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <maxGroups>3</maxGroups>
      </associationCommandClass>
    </entry>
    <entry>
      <commandClass>POWERLEVEL</commandClass>
      <powerLevelCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <powerLevel>0</powerLevel>
        <powerTimeout>0</powerTimeout>
      </powerLevelCommandClass>
    </entry>
    <entry>
      <commandClass>FIRMWARE_UPDATE_MD</commandClass>
      <firmwareUpdateCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
      </firmwareUpdateCommandClass>
    </entry>
    <entry>
      <commandClass>CONFIGURATION</commandClass>
      <configurationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <configParameters/>
      </configurationCommandClass>
    </entry>
    <entry>
      <commandClass>ZWAVE_PLUS_INFO</commandClass>
      <zwavePlusCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <zwPlusVersion>1</zwPlusVersion>
        <zwPlusRole>5</zwPlusRole>
        <zwPlusNodeType>0</zwPlusNodeType>
        <zwPlusDeviceType>0x600</zwPlusDeviceType>
        <zwPlusInstallerIcon>0x600</zwPlusInstallerIcon>
        <isGetSupported>true</isGetSupported>
      </zwavePlusCommandClass>
    </entry>
    <entry>
      <commandClass>MANUFACTURER_SPECIFIC</commandClass>
      <manufacturerSpecificCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <initSerialNumber>false</initSerialNumber>
        <deviceManufacturer>99</deviceManufacturer>
        <deviceType>18756</deviceType>
        <deviceId>12597</deviceId>
      </manufacturerSpecificCommandClass>
    </entry>
    <entry>
      <commandClass>SCENE_ACTUATOR_CONF</commandClass>
      <sceneActuatorConfigurationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </sceneActuatorConfigurationCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_ALL</commandClass>
      <switchAllCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
        <mode>SWITCH_ALL_INCLUDE_ON_OFF</mode>
      </switchAllCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_MULTILEVEL</commandClass>
      <multiLevelSwitchCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <isGetSupported>true</isGetSupported>
      </multiLevelSwitchCommandClass>
    </entry>
    <entry>
      <commandClass>NO_OPERATION</commandClass>
      <noOperationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </noOperationCommandClass>
    </entry>
    <entry>
      <commandClass>CRC_16_ENCAP</commandClass>
      <crc16EncapsulationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </crc16EncapsulationCommandClass>
    </entry>
    <entry>
      <commandClass>SCENE_ACTIVATION</commandClass>
      <sceneActivationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </sceneActivationCommandClass>
    </entry>
    <entry>
      <commandClass>ASSOCIATION_GROUP_INFO</commandClass>
      <associationGroupInfoCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <autoSubscribeGroups>
          <int>1</int>
        </autoSubscribeGroups>
      </associationGroupInfoCommandClass>
    </entry>
  </supportedCommandClasses>
  <securedCommandClasses/>
  <associationGroups>
    <entry>
      <int>1</int>
      <associationGroup>
        <index>1</index>
        <name>Lifeline</name>
        <profile1>0x0</profile1>
        <profile2>0x1</profile2>
        <commands>
          <commandClass>DEVICE_RESET_LOCALLY</commandClass>
          <commandClass>SWITCH_MULTILEVEL</commandClass>
        </commands>
        <associations/>
      </associationGroup>
    </entry>
    <entry>
      <int>2</int>
      <associationGroup>
        <index>2</index>
        <name>On/Off control</name>
        <profile1>0x20</profile1>
        <profile2>0x1</profile2>
        <commands>
          <commandClass>BASIC</commandClass>
        </commands>
        <associations/>
      </associationGroup>
    </entry>
    <entry>
      <int>3</int>
      <associationGroup>
        <index>3</index>
        <name>On/Off control</name>
        <profile1>0x20</profile1>
        <profile2>0x1</profile2>
        <commands>
          <commandClass>BASIC</commandClass>
        </commands>
        <associations/>
      </associationGroup>
    </entry>
  </associationGroups>
  <nodeNeighbors>
    <int>1</int>
    <int>8</int>
    <int>9</int>
  </nodeNeighbors>
  <lastSent>2017-04-22 17:04:50.612 UTC</lastSent>
  <lastReceived>2017-04-22 17:04:50.647 UTC</lastReceived>
</node>
I am using openHAB2 installed from the official release in the Debian repos,  I believe it is V2.0.0-1 and have all the add ons installed. Using the Aeotec Z-stick gen 5 as a controller
[GE_45613_manual.pdf](https://github.com/openhab/org.openhab.binding.zwave/files/950090/GE_45613_manual.pdf)

[OH2logs.zip](https://github.com/openhab/org.openhab.binding.zwave/files/950086/OH2logs.zip)

cdjackson, 2017-04-24:
Please see the database guide below -:
https://github.com/cdjackson/HABmin2/wiki/Z-Wave-Product-Database

Thanks.
Chris

theweez928, 2017-04-24:
I don't understand what you would like me to do next. I thought I was raising the issue with the required information. What else is needed?
Thank you,
Josh

cdjackson, 2017-04-24:
It would be appreciated if you could enter the information into the database - as per the link I provided. This is the quickest way to get this introduced. Alternatively, we will look to add it at some point, but it’s likely to take longer.

theweez928, 2017-04-26:
I was able to add the XML file to the database, but it is still showing as an unknown device. Is there something I need to do on my system to get the updated database?

cdjackson, 2017-04-26:
Once it is approved and merged into the binding (which probably hasn't happened) then you will need to update to the latest snapshot binding.

closed 2017-05-27 by cdjackson