openhab/org.openhab.binding.zwave #558: FGMS-001 Is currently marked as failed by the controller! But all works fine

brutevinch, 2017-04-24:
OH - 2.1
zwave - Last dev branch binding.

Manufacturer 010f Fibargroup
Type / ID 0800:1001
Firmware Version2.6
Basic ClassBASIC_TYPE_ROUTING_SLAVE
Generic ClassGENERIC_TYPE_SENSOR_BINARY
Specific ClassSPECIFIC_TYPE_ROUTING_SENSOR_BINARY
Using Security
Routing
Listening
Frequently Listening
Beaming
Last Wakeup TimeMonday, April 24th 2017, 10:31:51 2 hours ago
Neighbours 4 nodes 1, 3, 4, 5
Last Heal TimeMonday, April 24th 2017, 04:33:06 8 hours ago

In log i see often this:

12:26:24.466 [WARN ] [rialmessage.IsFailedNodeMessageClass] - NODE 2: Is currently marked as failed by the controller!
12:26:54.522 [WARN ] [rialmessage.IsFailedNodeMessageClass] - NODE 2: Is currently marked as failed by the controller!
12:27:24.571 [WARN ] [rialmessage.IsFailedNodeMessageClass] - NODE 2: Is currently marked as failed by the controller!
12:27:54.621 [WARN ] [rialmessage.IsFailedNodeMessageClass] - NODE 2: Is currently marked as failed by the controller!
12:28:24.652 [WARN ] [rialmessage.IsFailedNodeMessageClass] - NODE 2: Is currently marked as failed by the controller!

i know that it's battery device and it's always sleeps but is this message WARN?? can't understand(
BTW sensor with OH works fine motion and other data

cdjackson, 2017-04-24:
Sorry - I'm not sure what the issue is that you're reporting? If you just want to discuss how this works, then this should be done on the forum.

Please clearly state the bug that you want fixed.

brutevinch, 2017-04-24:
Node 2 is FGMS-001 and it's not failed now. it works well but in log i see NODE 2: Is currently marked as failed by the controller! i think it's bug.

cdjackson, 2017-04-24:
It's not a bug - the controller is reporting that it thinks the device is failed - this is the state from the controller.

closed 2017-04-24 by brutevinch