openhab/org.openhab.binding.zwave #560: Aeotec ZW095 Home Energy Meter Gen5 doesn't work properly

twisniew1976, 2017-05-04:
* OH2 Version: 2.1.0 SNAPSHOT
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): SNAPSHOT: May 1, 2017 12:19 PM

* Device name: Aeotec ZW095 Home Energy Meter Gen5
* Device version: ZW095-C

* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/375

* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.
[node16.xml.zip](https://github.com/openhab/org.openhab.binding.zwave/files/977520/node16.xml.zip)

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.
> 2017-05-03 18:02:41.477 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 2: CRC check failed message contains AD 7D  but should be 43 B9

More description:
- Using default configuration ZW095 completely doesn't work. Problem is connected with CRC16 (perhaps):

System reports error:
> 2017-05-03 18:02:41.477 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 2: CRC check failed message contains AD 7D  but should be 43 B9

Disabling: Reporting CRC-16 Encapsulation Command by OpenZWave Control Panel enable ZW095 to work in OH2.

- OpenHAB 2 doesn't see all 3 Clamps - only 2 and whole device is available.

- Changing from OpenHAB any of parametrs: 4-6, 8-10 enabling Reporting CRC-16 Encapsulation Command (parameter 13)

cdjackson, 2017-05-04:
Please provide a full debug level log of the problem.

twisniew1976, 2017-05-04:
I've remove thing from org.eclipse.smarthome.core.thing.Thing.json and add device again (by user interface). After this operation all works properly.

closed 2017-05-04 by twisniew1976