openhab/org.openhab.binding.zwave #565: Change sensor_power description and unit from Energy Consumption (kWh) to Power (W)

michel53, 2017-05-08:
The actual sensor_power description in 
\org.openhab.binding.zwave-master\ESH-INF\thing\_channels.xml
is misleading.
It should be changed 
from 
    
    <channel-type id="sensor_power">
        <item-type>Number</item-type>
        <label>Energy Consumption</label>
        <description>Indicates the energy consumption (kWh)</description>
        <category>Energy</category>
        <state pattern="%.1f" readOnly="true">
        </state>
    </channel-type>
to
    
    <channel-type id="sensor_power">
        <item-type>Number</item-type>
        <label>Actual Power Consumption</label>
        <description>Indicates the actual power consumption (W)</description>
        <category>Energy</category>
        <state pattern="%.1f" readOnly="true">
        </state>
    </channel-type>

cdjackson, 2017-05-13:
The above change will change all current KWH sensors to indicate W which will definitely then be wrong.

This change can only be made after a total search and replace on all channels using sensor_power in the database to change them to something else (sensor_energy?)

michel53, 2017-05-15:
I made a quick search through org.openhab.binding.zwave-master\ESH-INF\thing\ and found 19 devices using sensor_power.
All of them are using sensor_power for
SENSOR_MULTILEVEL;type=POWER
none for energy.
Energy is only used with METER (METER;type=E_KWh).
So it should be safe to change the thing_channels.xml description, right?

michel53, 2017-05-15:
If I understand the Z-Wave spec documents right 
( http://z-wave.sigmadesigns.com/design-z-wave/z-wave-public-specification/
http://zwavepublic.com/specifications
),
there is no sensor type energy with a kWh scale, only sensor type power with W or kW scale.
Do you agree?

hlat, 2017-05-22:
As far as I could check it I found for energy (kWh) the METER command class, like michel53 state the sensor type is for power (Watt)

UweRieping, 2021-01-09:
Noticed that the error is still not corrected. I'm not sure if this is a problem of the Binding but at least the documentation of the Fibaro FRG222 
[https://v2.openhab.org/v2.3/addons/bindings/zwave/thing.html?manufacturer=fibaro&file=fgr222_24_24.html](url) 
should be changed from "Indicates the energy consumption (kWh)." to "Indicates the energy consumption (W)."

michel53, 2021-01-11:
To be correct, it should be changed from "Indicates the energy consumption (kWh)." to "Indicates the power consumption (W)."