openhab/org.openhab.binding.zwave #567: Polling (REFRESH command) over un-initialized thing

IOOOTARobby, 2017-05-12:
* OH2 Version: ESH 0.9.0b4, with just the ZWave binding installed
* Binding Version: 2.0.0

The problem arises when we start ESH while one of the devices (domitech LED light bulb) is not powered.
ESH detects that device as "ONLINE", while it's not.

We periodically poll all the powered devices with a REFRESH command (once in a minute): it works perfectly even with the bulb **if it is powered during ESH boot**.

Otherwise, the REFRESH is regularly deferred because the device results to be not initialized.
The check is performed at [this line](https://github.com/openhab/org.openhab.binding.zwave/blob/2e3b7a9eb8eeccfe5a48405110c89bf5a5a68668/src/main/java/org/openhab/binding/zwave/handler/ZWaveThingHandler.java#L365) of the ZWaveThingHandler class.

The bulb seems to never complete its initialization, thus the polling is always deferred and ESH state results erroneously "ONLINE".

Switching the bulb ON and then OFF normalizes the situation: then the polling acts as expected (ONLINE and OFFLINE state are correctly detected).

I don't know if it should be possible to poll a device that is not completely initialized, I guess it would be not a proper action.
It may have sense to put a DEVICE in INITIALIZING or UNITIALIZED state if the poll is deferred for that particular reason.

Here is the relevant part of the log, the full log is attached, together with the node's XML.
```
2017-05-12 16:34:19,759 | DEBUG | safeCall-7       | ZWaveThingHandler                | NODE 3: Command received zwave:device:zwavectrl:node3:switch_dimmer --> REFRESH
2017-05-12 16:34:19,795 | DEBUG | safeCall-7       | ZWaveThingHandler                | NODE 3: Polling intialised at 1800 seconds - start in 50 milliseconds.
2017-05-12 16:34:19,825 | DEBUG | H-thingHandler-1 | ZWaveThingHandler                | NODE 3: Polling...
2017-05-12 16:34:19,845 | DEBUG | H-thingHandler-1 | ZWaveThingHandler                | NODE 3: Polling deferred until initialisation complete
```

[Node XML file attached](https://github.com/openhab/org.openhab.binding.zwave/files/997398/node3.xml.zip)
[Full log attached](https://github.com/openhab/org.openhab.binding.zwave/files/997390/org.openhab.binding.zwave.log.zip)

cdjackson, 2017-05-12:
Wouldn’t it be better to find out why the device isn’t initialising? If the device isn’t fully initialised, then it probably won’t work properly, so polling it is not the only problem.

IOOOTARobby, 2017-05-12:
I try to be clearer.
This is the flow:

- the device is initialized and is working properly => the polling works perfectly
- I turn the power off => the device goes OFFLINE in ESH, great
- I restart ESH
- the device does not get initialized (it's turned off), but it results to be ONLINE in ESH
- the polling is deferred forever, so the device is kept ONLINE in ESH, but it's actually turned off

I think the device cannot be initialized until I turn the power on again, but at the same time it should not result as ONLINE in ESH.

Maybe I don't have the full picture, I will be glad if you could explain it to me, thanks a lot.