openhab/org.openhab.binding.zwave #569: Yale (Assa Abloy) YRD246 not in database

TheKorn2, 2017-05-15:
Added to the zwave database, hopefully time to just waive the magic wand.  ;)

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/594

cdjackson, 2017-05-15:
I'll try and remember to update the dev binding tonight ;)

TheKorn2, 2017-05-15:
What's weird is that the device in the database says support security = NO, then talks about security classes.  (It for sure does security, and in habmin it says it's doing security, so...  ?????  Maybe I'm not reading that right?)

cdjackson, 2017-05-15:
Yeah - I’m not 100% sure what that refers to. It’s something that’s read from the device and displayed in the data - it’s not used for anything though so don’t worry about it...

TheKorn2, 2017-05-16:
Wand has been waved, lock works great.  Thanks!

closed 2017-05-16 by TheKorn2