openhab/org.openhab.binding.zwave #577: Update database:  ZW095

TheKorn2, 2017-05-21:
Please update database for Aeon Labs ZW095.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/375

cdjackson, 2017-05-21:
This will be managed through the next database update.

closed 2017-05-21 by cdjackson

TheKorn2, 2017-05-21:
So...  only open tickets for new things?  Not necessary to open tickets for updated old things?

cdjackson, 2017-05-21:
I prefer actually to handle both through the database, but sometimes people don’t know about the online database so raise an issue first. In your case, you’ve already updated the database and requested the review in the database so there doesn’t seem any point in also raising issues as well.

I will likely do an update tonight...