openhab/org.openhab.binding.zwave #588: Problem with Aeotec AEOEZW090-C controller => Aeotec Key Fob ZW088 discoverd only as unknown device

caesar1111, 2017-06-07:
* OH2 Version: 2.0 (running on Raspberry Pi 3 with latest Raspbian Jessie with PIXEL 
Version: April 2017
Release date: 2017-04-10
Kernel version: 4.4)
* Binding Version: 2.0.0

If the problem is associated with a device, please provide the following -:

* Device name: Aeotec AEOEZW090-C controller & Aeotec Key Fob ZW088
* Device version: Controller: Z-Stick Gen5 AEOEZW090-C KeyFob: Gen5 (both German/Europe editions)
* Link to [database]:
Controller: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/265
KeyFob: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/87

Issue:
Offline pairing (z-stick button) works fine and both devices are indication a connection
Adding the KeyFob in HABmin is producing only a "Unknown Device".
I also had the same issue with a borrowed Devolo KeyFob:http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/341

Other Z-Wave Devices are recognized without any problem using Z-Stick:
like:
-Aeotec Multisens 6:http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/387
-Fibargroup Double Swith 2:http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/416
-nodon smart plug, discovered as ID-RF:http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/435

Thanks in advance for helping

Caesar

cdjackson, 2017-06-07:
We need to know the device type and ID for your device - this should be shown in HABmin.

caesar1111, 2017-06-07:
Hi,

sorry but I cant give you the values of HABmin since the controller and the KeyFob are not discovered correctly.
The controller should be according to your database:References (Type:Id)	0001:005A,0101:005A,0201:005A
but comes up in HABmin as:
[image]

and the KeyFob should be according to your database:References (Type:Id)	0001:0058,0101:0058,0201:0058
but comes up in HABmin as:
[image]

But as mentioned above, even if the controller is not detected completely, some devices come up fine like the multi sensor:
[image]
with 0002:0064

Thanks in advance
Caesar

cdjackson, 2017-06-07:
Ignore the controller - this doesn’t matter.

I suggest you discuss this on the forum. I suspect it’s highly likely that you just haven’t woken up the device as it’s a battery device, but if you discuss on the forum I’m sure people will be able to help. If there’s a bug, then we can come back here, or if the database needs updating then that can also be done.

caesar1111, 2017-06-08:
Ok
I was just asking here first, since the Device in your database is in status "requires review"
[image]
About waking up the battery device:
Since the Z-Stick is supporting only the "offline pairing" (you have to unplug the stick and use the button on the stick to pair to the device) you cant use the HABmin device discovery function to pair the KeyFob with the controller. But if you are doing the pairing offline, both devices are indicating a successful pairing with the designated led codes.
If you want to add the paired device to HABmin, you have to plug in the Z-Stick and since the pairing was done offline before, the Z-Stick not the KeyFob now provides the device information for HABmin. So there should be not issue with the KeyFob not being awake.
I also did a complete offline un-pairing and mad sure the KeyFob was awake when using the dicovery function in HABmin.

So I will go hunting in the forum whether someone had a similar problem with the Z-Stick and battery devices.
Thanks for the help so far.

Caesar

caesar1111, 2017-06-08:
..here is the manual for the Z-Stick explaining the "offline" inclusion
https://aeotec.freshdesk.com/support/solutions/articles/6000056439-z-stick-gen-5-user-manual-

cdjackson, 2017-06-08:
The device is included already in the binding. If you don’t see the device id etc, then this is probably because the device hasn’t been woken up - this doesn’t matter if the device is in the database or not - it will still show this information if the binding has been able to communicate with the device and you say that this isn’t showing so I think you need to wake up the device.

The inclusion method is not relevant - the binding doesn’t care if you included it with the button on the stick or through the binding.

caesar1111, 2017-06-09:
Hi,

thks, problem solved.
the forum gave me the right hint and as always the devil is in the detail and it the inclusion mode matters, at least timing wise (30sec. vs. 8sec.):
you were right with the being awake thing, but timing is absolutely crucial (see below).

On top there was an issue with the manual of the device since the new standard manual didnt tell you that the wake-up time for manual  wake-up is only 8sec. while the wake-up while inclusion is 30sec. (found the information only in an old standard manual)

So what finally worked for me:
1. unplug the Z-Stick, press the button on the Z-Stick to start inclusion (stick is flashing blue)
2. start the inclusion on the KeyFob by pressing short the learn button on the back (or if you are still in setup mode the Button 4 lower rigth on the front side) the inclusion will be indicated at the stick with a solid blue light
3. plug in the Z-Stick again and wait for it to be recognized (online)
4. (and here timing is essential) including the KeyFob using HABmin while the KeyFob is awake. To wake up the KeyFob manually you have to press the learn button on the back for 3 seconds. But it only wakes up for 8 sec. So the sequence hast to be exactly:
=>wake-up KeyFob
=>start Z-Wave inclusion on HABmin within 8 sec.
=>wait for the KeyFob to be discovered as Thing: ZW088 4 button keyfob - Gen 5
If the device is still coming up as unknown or, what happened for me, came up as a other device, just delete the Thing again and redo the sequence.
I had to do it multiple times before the inclusion was done right
So if you wake up the KeyFob after starting the inclusion in HABmin or if you are to slow starting the inclusion in HABmin after waking up KeyFob, it will always come up as unknown device

So thanks for fast support.

closed 2017-06-10 by cdjackson