openhab/org.openhab.binding.zwave #592: Unable to get item details for Jasco ZW4003 In-Wall Smart Switch

mrbsayles, 2017-06-08:
In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version: 2.0
[openhab.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1062646/openhab.log.txt)

* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):

If the problem is associated with a device, please provide the following -:

* Device name: In-Wall smart switch (Jasco ZW4003)
* Device version: 5.0

Properties as shown:
zwave_class_basic	ROUTING_SLAVE
zwave_class_generic	BINARY_SWITCH
zwave_frequent	false
zwave_neighbours	1
zwave_version	5.20
zwave_listening	true
zwave_plus_devicetype	UNKNOWN_TYPE
zwave_deviceid	12343
zwave_nodeid	6
zwave_routing	true
zwave_beaming	true
zwave_class_specific	POWER_SWITCH_BINARY
zwave_manufacturer	99
zwave_devicetype	18770

* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list):
* If the device is not in the database, provide a link to the manual. https://byjasco.com/HEP-ET/00000-19999/14292
[OpenHab log file.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1062569/OpenHab.log.file.txt)

* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

cdjackson, 2017-06-09:
Please ensure you are using the latest version (ie snapshot). This should solve the issue.

For reference, database guide is below, but I think the device is already available.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

closed 2017-06-09 by cdjackson

mrbsayles, 2017-06-09:
Thanks! After updating to , the device is indeed found.,