openhab/org.openhab.binding.zwave #600: Secure Controls SRT 323 Thermostat without association group and configuration parameters

pjramiro, 2017-06-14:
Hi, I'm not sure if this is is the place for this issue.
I think that there is a problem/mistake with the Secure Controls SRT 323 Thing, because in HABmin, in Things --> Configuration, doesn't appears any "Configuration Parameters" neither "Association Groups". In the database, the https://github.com/openhab/org.openhab.binding.zwave/blob/master/ESH-INF/thing/horstmann_srt323_0_0.xml database doesn't appears this info too. https://github.com/openhab/org.openhab.binding.zwave/blob/master/ESH-INF/thing/horstmann_srt323_0_0.xml.
In the manufacturer instructions appears configuration parameters and association groups and I think that is imposible to work correctly with SRT 323 in openhab, without this parameters .

* OH2 Version: 2.0.0-1
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): Z-Wave Binding 2.0.0
* Device name: SRT323 Secure Z-Wave Room Thermostat http://products.z-wavealliance.org/products/1241
* Link to database: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/466

[z_wave_secure_srt323_room_thermostat_with_lcd_display_product_sheet.pdf](https://github.com/openhab/org.openhab.binding.zwave/files/1074742/z_wave_secure_srt323_room_thermostat_with_lcd_display_product_sheet.pdf)

[z_wave_secure_srt323_room_thermostat_with_lcd_display_user_and_installation_manual.pdf](https://github.com/openhab/org.openhab.binding.zwave/files/1074743/z_wave_secure_srt323_room_thermostat_with_lcd_display_user_and_installation_manual.pdf).

Thank you very much.

cdjackson, 2017-06-14:
Currently there is no configuration defined in the database so please feel free to add this. There is information on doing this here -:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

pjramiro, 2017-06-14:
Ok, I am new in this world (openhab and z wave), but I will try to do it in a few days. 
Thank you.

pjramiro, 2017-06-16:
Hi, I have readed your guides, I have looked other xml zwave examples (even srt323 openzawe xml), I have reviewed manufacturer instructions and finally, I have a new xml version (I attach it).

This is a summary from the manufacturer instructions in relation with my changes:

"The following association groups are supported:
Command Class Group 1 - Lifeline. The node associated to this group will receive messages from all the other groups, whether or not that group has itself been
associated. Only one node may be associated to this group.
Group 2 - Nodes to receive Thermostat Operating State reports.
Group 3 - Nodes to receive unsolicited Low Battery Warnings.
Group 4 - Nodes to receive unsolicited Thermostat Set Point Reports.
Group 5 - Nodes to receive unsolicited Multilevel Sensor Reports.
Group 1 contains a maximum of 1 node. Each of the groups 2,3,4 and 5 may contain a maximum of 4 nodes."
"The unit supports 3 single byte configuration Command Class parameters for the temperature sensor in the range 1-3 respectively.
Configuration Parameter Number 1, Default = 0xFF 
0x00 - 0x7F Disables the temperature sensor.
0x80 - 0xFF Enables temperature sensor.

Configuration Parameter Number 2, Default = 0
0x00 - 0x7F Celsius.
0x80 - 0xFF Fahrenheit.

Configuration Parameter Number 3, Default =
10 (1.0˚C)
1 to 100 - Delta Temperature in 0.1˚C steps"

As it is my first "xml thing" and I am a newbie, how could I test my xml file proposal? Can you or any other member of the community review it? And when the file will be reviewed, do you upload it directly or give me permissions and I upload it?
Thank you so much for everything.

[horstmann_srt323_0_0_v1.zip](https://github.com/openhab/org.openhab.binding.zwave/files/1081208/horstmann_srt323_0_0_v1.zip)

cdjackson, 2017-06-16:
Updates need to be made directly in the database editor online.

pjramiro, 2017-06-16:
Ok, can you give me access rights or you prefer to upload the file by yourself?. And can you take a look to the file?
Thank you very much for your effort.

cdjackson, 2017-06-16:
Sure - let me know your user name if you've registered and I can do this. You can't upload a file - just use the editor.

pjramiro, 2017-06-16:
pjramiro
Thanks!

cdjackson, 2017-06-16:
Done. Thanks. Any problems, just yell...

pjramiro, 2017-06-17:
Hi, reviewing and editing other database devices for understand how database works, I don't understand how but I submitted in error a review for hrt4-zw. Now I am trying to adapt xml file (configuration parameters) to database editor for srt323 device.
Excuse me.

pjramiro, 2017-06-17:
Hello, I finished editing the SRT323 device in the database, but I have doubts in relation with the field "Controller" in "Association Groups". The manufacturer says: "The following association groups are supported:
Command Class Group 1 - Lifeline. The node associated to this group will receive messages from all the other groups, whether or not that group has itself been
associated. Only one node may be associated to this group.
Group 2 - Nodes to receive Thermostat Operating State reports.
Group 3 - Nodes to receive unsolicited Low Battery Warnings.
Group 4 - Nodes to receive unsolicited Thermostat Set Point Reports.
Group 5 - Nodes to receive unsolicited Multilevel Sensor Reports.
Group 1 contains a maximum of 1 node. Each of the groups 2,3,4 and 5 may contain a maximum of 4 nodes."

Do I have to check the field "Controller" in all the "Association Groups"?
Thank you.

cdjackson, 2017-06-17:
Hi,
All looks ok. I removed the controller check from the groups other than the lifeline group. If you find it’s needed we can look to add the others, but normally the lifeline should be enough. The problem with adding all the others is that you can get multiple messages which can sometimes cause some problems.

I’ll update the binding tomorrow hopefully.

Thanks.

Cheers
Chris

pjramiro, 2017-06-20:
Hi, I have upgraded to the snapshot version and works fine. I modified the association group and other parameters in habmin (I have not had no time to test in PaperUI) and now the srt323 inmediatly sends to the controller set points changes and the relay status changes.
I will test it these weeks and I will send the feedback.
Thank you ver much. 
Cheers.

closed 2017-06-24 by cdjackson