openhab/org.openhab.binding.zwave #603: NPE in ZWaveBatteryConverter

martinvw, 2017-06-17:
* OH2 Version: 2.1.0-SNAPSHOT
* Binding Version: zwave:2.1.0.201705311713

If the problem is associated with a device, please provide the following -:

* Link to db: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/401

```
2017-06-17 08:02:07.798 [WARN ] [ommandclass.ZWaveBatteryCommandClass] - NODE 5: Battery state unknown (108)!
2017-06-17 08:02:07.798 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.NullPointerException
        at org.openhab.binding.zwave.internal.converter.ZWaveBatteryConverter.handleEvent(ZWaveBatteryConverter.java:71)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.ZWaveIncomingEvent(ZWaveThingHandler.java:1099)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveBatteryCommandClass.handleApplicationCommandRequest(ZWaveBatteryCommandClass.java:114)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:119)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[217:org.openhab.binding.zwave:2.1.0.201705311713]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[217:org.openhab.binding.zwave:2.1.0.201705311713]
```

martinvw, 2017-06-17:
I also get it for some other neocam devices, did they do the batter calculation overly simplified and do they pass out values over 100%?

cdjackson, 2017-06-17:
If you can provide a log with the received message it would be great.  Actually, I just need the message itself, so a single line showing the hex data from the received message would be really useful.

martinvw, 2017-06-19:
Did not yet check the code before and I am working remote so cannot fetch / enable the logging, but this looks fishy enough to me:

```java
// If the battery level is outside bounds, then we don't know what's up!
if (batteryLevel < 0 || batteryLevel > 100) {
    logger.warn("NODE {}: Battery state unknown ({})!", this.getNode().getNodeId(), batteryLevel);
    batteryLevel = null;
}
ZWaveCommandClassValueEvent zEvent = new ZWaveCommandClassValueEvent(this.getNode().getNodeId(), endpoint, this.getCommandClass(), batteryLevel);
```
[line 107 @ ZWaveBatteryCommandClass](https://github.com/openhab/org.openhab.binding.zwave/blob/01264939c156ed6e5d9e7e89ac511c6100eeedee/src/main/java/org/openhab/binding/zwave/internal/protocol/commandclass/ZWaveBatteryCommandClass.java#L107)

The logging states that the value is over 100, in that case the `batteryLevel` is set to `null` and still passed on to the [ZWaveBatteryConverter @ line 71](https://github.com/openhab/org.openhab.binding.zwave/blob/01264939c156ed6e5d9e7e89ac511c6100eeedee/src/main/java/org/openhab/binding/zwave/internal/converter/ZWaveBatteryConverter.java#L71).

I think that the actual null pointer is caused by a nice combination of auto-boxing and widening. Because the Integer value which we forward to the `DecimalType` is auto-unboxed to an `int` with the goal to construct an `long` out of it. Unfortunately the auto-unboxing already fails with an NPE.

Just be sure, do you still need logging?

cdjackson, 2017-06-19:
> Just be sure, do you still need logging?

Yes please - the fix is easy, but I'm also trying to add tests every time I fix something so I'd like to get a 'real' packet to use in the tests.

cdjackson, 2017-06-19:
If it's too hard to get the log I'll generate a fake packet myself, but if possible, I prefer to use 'live' data.

martinvw, 2017-06-19:
No I just need to go home tonight, but my wife expects that anyway after a day of work ;-)

martinvw, 2017-06-19:
```
2017-06-07 19:38:46.063 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 00 04 00 05 03 80 03 6C 1B
2017-06-07 19:38:46.063 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-06-07 19:38:46.063 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 04 00 05 03 80 03 6C 1B
2017-06-07 19:38:46.063 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 09 00 04 00 05 03 80 03 6C 1B
2017-06-07 19:38:46.063 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 05 03 80 03 6C
2017-06-07 19:38:46.063 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Application Command Request (ALIVE:DONE)
2017-06-07 19:38:46.063 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 5: Starting initialisation from DONE
2017-06-07 19:38:46.063 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@233f49a3 already registered
2017-06-07 19:38:46.063 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Incoming command class BATTERY
2017-06-07 19:38:46.064 [DEBUG] [ommandclass.ZWaveBatteryCommandClass] - NODE 5: Received Battery Request
2017-06-07 19:38:46.064 [DEBUG] [ommandclass.ZWaveBatteryCommandClass] - NODE 5: Battery report value = 108
2017-06-07 19:38:46.064 [WARN ] [ommandclass.ZWaveBatteryCommandClass] - NODE 5: Battery state unknown (108)!
2017-06-07 19:38:46.064 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveCommandClassValueEvent
2017-06-07 19:38:46.064 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveCommandClassValueEvent2017-06-07 19:38:46.064 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 0, command class = BATTERY, value = null
2017-06-07 19:38:46.064 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.NullPointerException

```

cdjackson, 2017-06-19:
Thanks. I’ve made a change so let me know if this fixes the issue.

Cheers
Chris

martinvw, 2017-06-22:
I somehow missed your message, but luckily I updated and just checked it out :-)

[image]

Given your code change I would not expect this...

     223 | Active   |  80 | 2.1.0.201706221855     | ZWave Binding

cdjackson, 2017-06-22:
Why wouldn't you expect this? It seems ok to me - no NPE and a valid value?

martinvw, 2017-06-22:
It's the 112% I read your code as it should be converted to 100%

cdjackson, 2017-06-22:
Ah - I didn't see that. I thought you were showing the 100% value at the bottom.

Are you sure this is actually updating - it's not just restoring from persistence or something? I added tests for this, and values over 100% get converted to 100% so I'm not sure how 112% can show up.

martinvw, 2017-06-22:
The project with the tests is fine :-P

But the one without the tests not, see the link in my edited message:

https://github.com/openhab/org.openhab.binding.zwave/pull/609/files#diff-548402f35e072a7d90a29ff908d2e2e6R77

cdjackson, 2017-06-22:
Sorry - I don't know what you mean? The two versions are the same I think, and I still don't see what you think is wrong. Can you be explicit please.

The command class should return the value directly - so 112%. This is fine. It's the converter that then checks if the value is >100 or <0 and limits appropriately. Can you tell me what's wrong (sorry for being a bit slow).

cdjackson, 2017-06-22:
Ah - sorry, just seen the issue....

martinvw, 2017-06-22:
On the line I reference you use the original `event.getValue()` and not the `value` which was just corrected.

martinvw, 2017-06-22:
Thanks for your time!

cdjackson, 2017-06-22:
I've made another update - hopefully fixed this time ;)

Thanks.

cdjackson, 2017-06-24:
This should be closed - any problems feel free to reopen.

closed 2017-06-24 by cdjackson

martinvw, 2017-06-25:
Yes, indeed thanks! I saw no NPE and value of 100%