openhab/org.openhab.binding.zwave #610: No sensor_binary channel for fibaro_fgms001_0_0.xml any more :/

philomatic, 2017-06-21:
The motion sensor:

https://github.com/openhab/org.openhab.binding.zwave/blob/master/ESH-INF/thing/fibaro_fgms001_0_0.xml

doesn't have a sensor_binary for motion detection any more :(
It "disappeared" in:

https://github.com/openhab/org.openhab.binding.zwave/commit/7786415f330ae0c5f02f95a72942a9320e903c21#diff-bc5b461b7f9b5171536c91a02f5226d8

There's now an alarm_motion channel, but this one doesn't trigger on motion detection.
(I guess only when in alarm motion detection mode).

Log:
[fgms.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1091890/fgms.txt)

* Binding Version: master

* Device name: fibaro_fgms001_00_000
* Device version: All up to 2.8
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/115)

cdjackson, 2017-06-21:
> 7786415#diff-bc5b461b7f9b5171536c91a02f5226d8 <https://github.com/openhab/org.openhab.binding.zwave/commit/7786415f330ae0c5f02f95a72942a9320e903c21#diff-bc5b461b7f9b5171536c91a02f5226d8>

I’m guessing there might be parameter to configure what to send. The notification class is generally preferred.

philomatic, 2017-06-23:
Ok, I've checked this again with a fresh install and a new Z-Wave network and I'm sorry for the confusion:

Seems like https://github.com/openhab/org.openhab.binding.zwave/commit/2620be3dda2fc9424d99b62ab602a98ee7a6d272#diff-bc5b461b7f9b5171536c91a02f5226d8 fixed this.

I've had https://github.com/openhab/org.openhab.binding.zwave/commit/4006d8bebc94a09ce9d9979bb5a83616426d38d2 as my binding version because my osgi bundle cache didnt't correctly update.

Following the manual of this sensor, there's no special parameter for setting motion as an alarm frame. However, Z-Way Server software has a separate motion alarm channel which confused me (and does nothing, maybe its a misspelling and should be the tamper alarm when configured).

Solved an closed.

closed 2017-06-23 by philomatic