openhab/org.openhab.binding.zwave #612: FGDW-002 - Fibaro Door/Window Sensor

petzlux, 2017-06-23:
* OH2 2.1.0.201706222153 
* FGDW-002 Fibaro Door/Window Sensor
* EU 3.2 ZW5

Just bought 3 of these, and I can include them fine and configure them, but the values don't get updated. This is the case for the Door Open/Close Channel, as well as the Temperature Channel. 

````
2017-06-23 21:56:01.710 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 13 00 04 00 21 0D 56 01 71 05 00 00 00 FF 06 16 00 FF C5 32
2017-06-23 21:56:01.739 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-06-23 21:56:01.741 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 13 00 04 00 21 0D 56 01 71 05 00 00 00 FF 06 16 00 FF C5 32
2017-06-23 21:56:01.743 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 13 00 04 00 21 0D 56 01 71 05 00 00 00 FF 06 16 00 FF C5 32
2017-06-23 21:56:01.745 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 21 0D 56 01 71 05 00 00 00 FF 06 16 00 FF C5
2017-06-23 21:56:01.746 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 33: Application Command Request (ALIVE:DONE)
2017-06-23 21:56:01.747 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 33: Starting initialisation from DONE
2017-06-23 21:56:01.748 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@1cde97a already registered
2017-06-23 21:56:01.749 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 33: Incoming command class CRC_16_ENCAP
2017-06-23 21:56:01.750 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 33: Received CRC 16 Encapsulation Request
2017-06-23 21:56:01.751 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 33: Calling handleApplicationCommandRequest.
2017-06-23 21:56:01.752 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 33: Received ALARM command V5
2017-06-23 21:56:01.753 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 33: Process NOTIFICATION_REPORT V5
2017-06-23 21:56:01.754 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 33: NOTIFICATION report - 0 = 0, event=22, status=255
2017-06-23 21:56:01.755 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 33: Alarm Type = ACCESS_CONTROL (0)
2017-06-23 21:56:01.755 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
2017-06-23 21:56:01.756 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 33: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2017-06-23 21:56:01.757 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 33: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
2017-06-23 21:56:01.759 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2017-06-23 21:56:01.760 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2017-06-23 21:56:01.761 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 22, type OnOffType
2017-06-23 21:56:01.761 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 33: Updating channel state zwave:device:15bbb5fe595:node33:alarm_access to ON [OnOffType]
2017-06-23 21:56:01.762 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.IllegalStateException: Could not update state, because callback is missing
	at org.eclipse.smarthome.core.thing.binding.BaseThingHandler.updateState(BaseThingHandler.java:267)[105:org.eclipse.smarthome.core.thing:0.9.0.201706211517]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.ZWaveIncomingEvent(ZWaveThingHandler.java:1104)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass.processNotificationReport(ZWaveAlarmCommandClass.java:194)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass.handleApplicationCommandRequest(ZWaveAlarmCommandClass.java:124)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveCRC16EncapsulationCommandClass.handleCRC16EncapResponse(ZWaveCRC16EncapsulationCommandClass.java:138)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveCRC16EncapsulationCommandClass.handleApplicationCommandRequest(ZWaveCRC16EncapsulationCommandClass.java:73)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:119)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[220:org.openhab.binding.zwave:2.1.0.201706222153]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[220:org.openhab.binding.zwave:2.1.0.201706222153]
````

hasterhorb, 2017-08-16:
Here are two threads having the same problem:

https://community.openhab.org/t/fgdw-002-door-sensor-not-showing-status/30979
https://community.openhab.org/t/fibaro-door-windows-sensor-2-fgdw-002-problems/29905/11

cdjackson, 2017-12-30:
Closing duplicate of #716

closed 2017-12-30 by cdjackson