openhab/org.openhab.binding.zwave #620: Rollershutter invert percent option does not work for Basic command class

sihui62, 2017-06-25:
OH2 Version: Snapshot build 964
Binding Version: dev branch 2.1.0.201706242319
Device name: [Fibaro FGRM222 Rollershutter 2](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/413)
Debug log: https://gist.githubusercontent.com/sihui62/8173f08da9335c36a66c9fe6023157df/raw/d0790a1fd1d166934e31f34eb9721d6071664695/zwave_node32.log

Related Community forum thread: https://community.openhab.org/t/fibaro-roller-shutter-fgrm-222-with-venetian-blinds-lamellas-position/9396/133

Maybe fix is also needed for invert control option? I don't use this channel so cannot verify.

sihui62, 2017-07-03:
Workaround: removing BASIC messages from SWITCH_MULTILEVEL command class through database update solves the problem.
Is this workaround is a permanent fix the issue can be closed.
Thx.

ThomasBrodkorb, 2017-10-04:
Is there a step-by-step guidance on how to "remove BASIC messages from SWITCH_MULTILEVEL command class through database update" ?

I have no idea what to do.

sihui62, 2017-10-04:
> I have no idea what to do.

You don't have to do anything. The changes to the database were already made a few month ago.
All you have to do is using a recent snapshot or development zwave binding version:

snapshot: https://openhab.ci.cloudbees.com/job/openHAB2-Bundles/lastSuccessfulBuild/org.openhab.binding$org.openhab.binding.zwave/

development: https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653

ThomasBrodkorb, 2017-10-04:
Thanks, I just did the following and it seems to be working now:
* uninstalled the ZWave binding from Paper UI
* copied the zwave binding (org.openhab.binding.zwave-2.1.0-SNAPSHOT.jar) to \\pi3\openHAB-share\openhab2-addons
* removed the ZWAVE things in paper ui and readded them.
* Now I do have the additional channels for "Shutter Position" and "Lamella Position" available - and they seem to work

cdjackson, 2017-12-30:
Seems this is resolved from the above statement.

closed 2017-12-30 by cdjackson