openhab/org.openhab.binding.zwave #630: Philio PSP05 Motion Sensor - Add configuration definitions to database

mfkeenan, 2017-07-02:
The current definition of the Philio PSP05 Motion Sensor does not appear to have many of the configuration parameters allowed by the device in the database.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/600

Manual is attached and configuration parameters extracted to image below for convenience.
[image]

Cheers
Mark

[z_wave_philio_motion_sensor_psp05_gen5_manual.pdf](https://github.com/openhab/org.openhab.binding.zwave/files/1117856/z_wave_philio_motion_sensor_psp05_gen5_manual.pdf)

cdjackson, 2017-07-02:
Please consider updating the database to add the parameters. This will be much quicker than waiting for someone else to do it.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

sihui62, 2017-07-02:
Adding them by yourself is simple:

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

Open an account, email Chris for edit rights and you are good to go.

mfkeenan, 2017-07-02:
Thanks guys. Be happy to do it myself. Just created and account and mailed Chris for access.

cdjackson, 2017-08-08:
Closing this as it should be managed directly into the database...

closed 2017-08-08 by cdjackson