openhab/org.openhab.binding.zwave #631: Race condition appeaers when binding is running for 24+ hours

holmes, 2017-07-02:
I've got 6 ZWave switches throughout my house (start small) and am running into something like a race condition after the binding has been running for 24 hours. The switches are the Homeseer ones that report instant updates and support single/double/triple tap.

I'm assuming it's a race condition because of a few things:
1. It only happens in the mornings, and is fixed after a restart
1. If I leave logging at INFO, it happens 100%. DEBUG it happens 50%, TRACE logging it works perfectly.
  
### My Setup
When the switch is tapped on, it queries the current state and then determines what to do:

* If the light is off, it turns on to 35%. 
* If the light is at 35%, it then turns to 85%.
* If the light is at 85%, it turns down to 35%.
* If you tap the switch off it turns it off. 

This logic is implemented via rules and the percentage values come from a local webservice.

This works flawlessly for the first 24 hours. But then it starts acting weird. The initial tap on works - it goes to 35%. But the subsequent taps don't. This morning I'm seeing it in my Family Room and the Nursery. Yesterday it was in the Family Room/Kitchen/Dining Room. So it's not consistent based on switch.

I've got logs, and it looks like a race condition. The first tap always works flawlessly, the subsequent taps are the issues. I'll include the full logs, but here's a breakdown from when the second tap starts.

## [Working Lights - Kitchen](https://gist.github.com/holmes/8650fcbfa40c154a30133102182ccc95)
https://gist.github.com/holmes/8650fcbfa40c154a30133102182ccc95

1. Starting from 08:04:40
1. Scene Notification & update that scene 1 was tapped
1. Rule is triggered to get current state which sends a status request?
1. Receive the current state via SWITCH_MULTILEVEL_REPORT
1. Updates internal state to 31 (current state)
1. Send Command to change light to 85 (new state)
1. Receive ACK 

## [Failing Lights - Family Room](https://gist.github.com/holmes/a223ccf12ffe739e2e294644c385fa31)
https://gist.github.com/holmes/a223ccf12ffe739e2e294644c385fa31

1. Starting from 08:07:24
1. Scene Notification & update that scene 1 was tapped
1. Rule is triggered to get current state which sends a status request?
1. **Send Command to change light to 85 (new state)**
1. **Receive the current state via SWITCH_MULTILEVEL_REPORT (31, the old state)**
1. **Updates internal state to 31 (current state)**
1. **Ack Received**

So this one updates out of order and leaves OpenHab confused about what the current state actually is.

My current solution is to restart the binding every morning. Clearly I'd like to find a fix but it's hard to debug anything since this only happens after the binding has been running for a while. I've looked at the code very briefly but thought I'd post here in case someone had a clue.

Originally posted in the community: https://community.openhab.org/t/zwave-race-conditions-at-24-hrs/30855

cdjackson, 2017-07-02:
This is already being discussed elsewhere.

closed 2017-07-02 by cdjackson

holmes, 2017-07-02:
I spent this morning trying to understand how the protocol/code works. I'm still not 100% I understand it.

Let me know if there's a spot I can add some logging to that will help. I've created/built/deployed other bindings so I can do that here.

cdjackson, 2017-07-02:
Please, let's discuss in a single place, otherwise it's going to cause problems.

My guess at the moment is that you're just lucky it works for the first 24 hours and you're trying to do something that the system can't do and I'd suggest to add some delays in your rules.