openhab/org.openhab.binding.zwave #633: Multisensor PST02-A motion detection not working correctly.

tomekz28, 2017-07-08:
* OH2 Version:2.1
* Binding Version: 2.1

If the problem is associated with a device, please provide the following -:

* Device name: Multisensor PST02-A
* Device version:
* Link:http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/168

I have 1 (or potentially 2 issues) trying to get the sensor to send motion detection data. Illuminance, temperature works fine.

Issue numer one: seems to be an issue with Configuration Parameter Operation Mode, it is defaulted to 0, when trying to set it to 60  (that is what I had in z-wave 1.8 binding with openhab1) it returns ERROR 400 - bad request. Setting it to preset value of 10 also returns this error.  A short term workaround I did was to configure the sensor in openhab1, and then boot up OH2. The sensor started to work correctly and detect motion and change state. However, once the wake-up happened, the config from OH2 z-wave binding 2.1 got loaded up the Operation mode got reset to 0 and the motion detection stopped working.

Issue number 2: Even when the sensor was detecting motion in OH2, the paperui did not show it. Could it be because the discovery returned the Motion Sensor channel as Switch? In the OH1, I had it set to Contact instead, and it was working fine there.  IN OH2 resetting the linked item to Contact did not do anything.

XML: 
            <int>5</int>
            <configurationParameter>
              <index>5</index>
              <size>1</size>
              <value>0</value>
              <readOnly>false</readOnly>
              <writeOnly>false</writeOnly>
            </configurationParameter>

cdjackson, 2017-07-08:
For point 1, what software are you using? HABmin?
For point 2, you should use a contact on the door sensor - not a switch - this is required for the sensor_door channel.

I would strongly suggest to discuss these issues on the forum - you will have a much wider audience of people who can help. If there is a problem with the binding, then we can open an issue.

tomekz28, 2017-07-09:
Hi Chris,

Regarding point 1 -> I am using the OH2.1 version of Habmin.
Regarding point 2 -> After the discovery, several channels showed up
including the Door/Window Sensor of type Contact (but i am not using this
channel) and the Motion Sensor channel with type Switch (which I believe
should have been a Contact). When creating an Item link for the Motion
Sensor channel, the only type could be selected was "Switch". After
linking, I could go in into the item and change it from switch to contact,
but the Channel type of the Motion Sensor remained as Switch, while the
link changed to contact. Sorry if it is a bit confusing. :)

I will post the issues on the forum tomorrow, it is past 2 am here and I
need to call it a night. Thank you for helping me out. :)

- Tom

cdjackson, 2017-12-30:
> I will post the issues on the forum tomorrow, it is past 2 am here and I
need to call it a night. Thank you for helping me out. :)

Assuming this was discussed on the forum so closing. If there was a bug to be resolved then please update the details.

closed 2017-12-30 by cdjackson