openhab/org.openhab.binding.zwave #637: Create an option to set up a separate BASIC command class instead of mapping it to the specific one

roher-roher, 2017-07-13:
* OH2 Version: 2.1.0 stable
* Binding Version: 2.1.0 stable

* Device name: TKB Home TZ66D Dual Paddle Wall Switch; TKB Home TZ65D Dual Paddle Wall Dimmer
* Device version: ?
* Link to database: [TZ66D](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/157); [TZ65D](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/155)
* XML files attached:
[TZ65D.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1146362/TZ65D.txt)
[TZ66D.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1146361/TZ66D.txt)

Please see problem description in the following thread:
https://community.openhab.org/t/z-wave-basic-command-class-channel-is-missing/31076

roher-roher, 2017-12-02:
Hi, any news on this?

cdjackson, 2017-12-30:
This will be added into the dev binding in the coming day or so.

roher-roher, 2018-07-27:
Hi, any news? What does "Awaiting feedback" label mean?

cdjackson, 2018-07-27:
IT was added to the TZ55 only. I've just added it to the TZ65.

roher-roher, 2018-10-28:
Hi, sorry, cannot figure out which version of binding I should take for testing.

cdjackson, 2018-10-28:
Snapshot.

roher-roher, 2019-01-05:
Hi @cdjackson, Merry Christmas and Happy New Year!
Finally, while moving my Openhab 2 installation to another server, I have upgraded it to 2.4, and installed 2.5-SNAPSHOT Z-wave binding (downloaded from CI on 3rd of January), all nodes initialized successfully, but I still don't see basic channel available in PaperUI neither for TZ65D, nor for TZ66D devices. What am I doing wrong?

roher-roher, 2023-08-17:
Hi, any news on this issue?