openhab/org.openhab.binding.zwave #643: UNINITIALIZED - HANDLER_MISSING_ERROR when using textual thing configuration

brstein, 2017-07-25:
OH-Version = 2.1 (official)
ZWave-Binding = 2.1.0.201707192301 (devel-branch)
Configuration = ZStick Gen 5 + Multisensor 6 (node 8), nodes 3 and 4 are currently dead

**Problem:** 
When using textual thing configuration the multisensor (node_id=8) gets stuck in the state 
UNINITIALIZED - HANDLER_MISSING_ERROR, while the ZSticks comes up properly. If I do NOT use the textual configuration for the multisensor it will be discovered and initialized.

**Error-Log:**
> 2017-07-25 12:41:31.958 [INFO ] [ing.zwave.handler.ZWaveSerialHandler] - Connecting to serial port '/dev/ttyACM0'
> 2017-07-25 12:41:32.038 [INFO ] [ing.zwave.handler.ZWaveSerialHandler] - Serial port is initialized
> 2017-07-25 12:41:32.073 [INFO ] [ve.internal.protocol.ZWaveController] - Starting ZWave controller
> 2017-07-25 12:41:32.074 [INFO ] [ve.internal.protocol.ZWaveController] - ZWave timeout is set to 5000ms. Soft reset is false.
> 2017-07-25 12:41:35.257 [INFO ] [age.SerialApiGetInitDataMessageClass] - NODE 1: Node found
> 2017-07-25 12:41:35.258 [INFO ] [age.SerialApiGetInitDataMessageClass] - NODE 3: Node found
> 2017-07-25 12:41:35.258 [INFO ] [age.SerialApiGetInitDataMessageClass] - NODE 4: Node found
> 2017-07-25 12:41:35.259 [INFO ] [age.SerialApiGetInitDataMessageClass] - NODE 8: Node found
> 2017-07-25 12:41:35.260 [INFO ] [age.SerialApiGetInitDataMessageClass] - ZWave Controller using Controller API
> 2017-07-25 12:41:35.261 [INFO ] [age.SerialApiGetInitDataMessageClass] - ZWave Controller is Primary Controller
> 2017-07-25 12:41:35.261 [INFO ] [age.SerialApiGetInitDataMessageClass] - ------------Number of Nodes Found Registered to ZWave Controller------------
> 2017-07-25 12:41:35.262 [INFO ] [age.SerialApiGetInitDataMessageClass] - # Nodes = 4
> 2017-07-25 12:41:35.262 [INFO ] [age.SerialApiGetInitDataMessageClass] - ----------------------------------------------------------------------------
> 2017-07-25 12:41:35.607 [WARN ] [rialmessage.IsFailedNodeMessageClass] - NODE 4: Is currently marked as failed by the controller!
> 2017-07-25 12:41:35.615 [WARN ] [rialmessage.IsFailedNodeMessageClass] - NODE 3: Is currently marked as failed by the controller!
> 2017-07-25 12:41:41.482 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while calling thing handler factory 'org.openhab.binding.zwave.internal.ZWaveHandlerFactory@164b36a' 
java.util.concurrent.TimeoutException
> 	at java.util.concurrent.FutureTask.get(FutureTask.java:205)[:1.8.0_141]
> 	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:194)[98:org.eclipse.smarthome.core:0.9.0.b5]
> 	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:83)[98:org.eclipse.smarthome.core:0.9.0.b5]
> 	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:67)[98:org.eclipse.smarthome.core:0.9.0.b5]
> 	at org.eclipse.smarthome.core.thing.internal.ThingManager.doUnregisterHandler(ThingManager.java:795)[105:org.eclipse.smarthome.core.thing:0.9.0.b5]
> 	at org.eclipse.smarthome.core.thing.internal.ThingManager.unregisterHandler(ThingManager.java:785)[105:org.eclipse.smarthome.core.thing:0.9.0.b5]
> 	at org.eclipse.smarthome.core.thing.internal.ThingManager.unregisterAndDisposeHandler(ThingManager.java:866)[105:org.eclipse.smarthome.core.thing:0.9.0.b5]
> 	at org.eclipse.smarthome.core.thing.internal.ThingManager.access$17(ThingManager.java:860)[105:org.eclipse.smarthome.core.thing:0.9.0.b5]
> 	at org.eclipse.smarthome.core.thing.internal.ThingManager$3.run(ThingManager.java:289)[105:org.eclipse.smarthome.core.thing:0.9.0.b5]
> 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_141]
> 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_141]
> 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)[:1.8.0_141]
> 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)[:1.8.0_141]
> 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)[:1.8.0_141]
> 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)[:1.8.0_141]
> 	at java.lang.Thread.run(Thread.java:748)[:1.8.0_141]
> 2017-07-25 12:42:41.554 [ERROR] [ome.core.thing.internal.ThingManager] - Thing type migration failed for zwave:device:controller:node8. The handler deregistration did not complete within 60.000ms.
> 2017-07-25 12:42:46.560 [WARN ] [ome.core.thing.internal.ThingManager] - Initializing handler for thing 'zwave:device:controller:node8' takes more than 5000ms.
> 2017-07-25 12:42:46.613 [WARN ] [mon.registry.AbstractManagedProvider] - Could not update element with key zwave:device:controller:node8 in ManagedThingProvider, because it does not exists.

**Thing config used:**

> Bridge  zwave:serial_zstick:controller [ port="/dev/ttyACM0", controller_master="true", controller_suc="false", heal_enable="true" ,heal_time="3", controller_softreset="false", inclusion_mode="2", security_inclusionmode="0", controller_inclusiontimeout="30", security_networkkey="11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff 00" ]
>
> Thing zwave:device:controller:node8 "MyTestSensor" (zwave:serial_zstick:controller) [ node_id=8 ]

Despite this HANDLER_MISSING_ERROR, I can see the right properties of this node in the paper ui:

> zwave_class_basic | BASIC_TYPE_ROUTING_SLAVE
> zwave_class_generic | GENERIC_TYPE_SENSOR_MULTILEVEL
> zwave_neighbours | 1
> zwave_frequent | false
> zwave_version | 1.6
> zwave_listening | true
> zwave_plus_devicetype | NODE_TYPE_ZWAVEPLUS_NODE
> zwave_deviceid | 100
> zwave_nodeid | 8
> zwave_routing | true
> zwave_plus_roletype | ROLE_TYPE_SLAVE_SLEEPING_REPORTING
> zwave_beaming | true
> zwave_secure | false
> zwave_class_specific | SPECIFIC_TYPE_ROUTING_SENSOR_MULTILEVEL
> zwave_manufacturer | 134
> zwave_devicetype | 2

Sincerly,
   brstein

cdjackson, 2017-07-25:
I think it's most likely caused by your thing configuration - everything is defined as a string, but not everything is a string. For starters, the node ID is an integer.

I'd suggest to take a look at the following -:
https://community.openhab.org/t/zwave-manual-thing-configuration/26542

brstein, 2017-07-25:
Hello Chris,

sorry I do not understand, what you try to tell me with 

> everything is defined as a string, but not everything is a string. For starters, the node ID is an integer

I thought, that I already defined the node_id as integer by omitting the quotes (s. above). 

What I found out so far is: If I use the name which is assigned by the discovery (bold part below), I will receive the  HANDLER_MISSING_ERROR:

> Thing zwave:**device**:controller:node8 "MyTestSensor" (zwave:serial_zstick:controller) [ node_id=8 ]

If I use 

> Thing zwave:**aeon_zw100_00_000**:controller:node8 "MyTestSensor" (zwave:serial_zstick:controller) [ node_id=8 ]

as your linked document suggests, the device comes up and gets ONLINE, but it is also autodiscovered in the inbox with the name zwave:**device**:controller:node8.

Please be so kind to answer the following questions:

- What is the meaning of the thing "type id" being "device" when autodiscovered or "aeon_zw100_00_000" if provided manually?
- Why is this node autodiscovered for a seconde time when using the thing definition with "aeon_zw100_00_000"?
- How is the "application version" provided in the textual thing definition?

Thank you for your help,
  brstein

cdjackson, 2017-07-25:
> What is the meaning of the thing "type id" being "device" when autodiscovered or "aeon_zw100_00_000" if provided manually?

It isn't 'device' - it is a real type id just as you have here. You are thinking of the thing uid, not the thing type uid.

> Why is this node autodiscovered for a seconde time when using the thing definition with "aeon_zw100_00_000"?

At the moment this is how it is working. We can probably change this, but as is discsused on the forum, this is how it currently works.

> How is the "application version" provided in the textual thing definition?

Application version is not provided - it's a property of the device.

I don't think there is any bug here - it seems like it is working if things are going online. Can I suggest that we instead discuss this on the forum so that others can also benefit.

closed 2017-07-25 by cdjackson

brstein, 2017-07-25:
Hello Chris,

you are right it is a PEBCAC (Problem Exists Between Keyboard And Chair ;-)

I am currently struggling with the documentation.

Thank you for your quick help,
   brstein.

cdjackson, 2017-07-25:
:)

Please feel free to discuss on the forum and we can try and resolve any doc / etc issues...