openhab/org.openhab.binding.zwave #649: First Alert Zcombo database update

hansaya, 2017-08-04:
Device address is missing out of the database. Zcombo seems to be in the database but my new unit did not got recognized as the proper zwave device. 

* OH2 Version: Openhan2
* Device name: First alert ZCOMBO
* Device version: 2017 may 16
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/238

xml file that got generated:
[node4.zip](https://github.com/openhab/org.openhab.binding.zwave/files/1199018/node4.zip)

This is what I saw on the logs:

> 
> 2017-08-03 19:45:46.577 [InboxAddedEvent           ] - Discovery Result with UID 'zwave:device:f1150c69:node4' has been added.
> 2017-08-03 19:45:46.594 [BindingEvent              ] - org.openhab.binding.zwave.event.BindingEvent@1c2bb33
> 2017-08-03 19:46:16.589 [BindingEvent              ] - org.openhab.binding.zwave.event.BindingEvent@105517b
> 2017-08-03 20:23:04.354 [InboxRemovedEvent         ] - Discovery Result with UID 'zwave:device:f1150c69:node4' has been removed.
> 2017-08-03 20:23:04.360 [ThingAddedEvent           ] - Thing 'zwave:device:f1150c69:node4' has been added.
> 2017-08-03 20:23:04.429 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from UNINITIALIZED to INITIALIZING
> 2017-08-03 20:23:04.436 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
> 2017-08-03 20:23:04.440 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to ONLINE
> 2017-08-03 20:30:28.559 [ThingUpdatedEvent         ] - Thing 'zwave:device:f1150c69:node4' has been updated.
> 2017-08-03 20:30:28.561 [ConfigStatusInfoEvent     ] - ConfigStatusInfo [configStatusMessages=[]]
> 2017-08-03 20:30:30.146 [ThingUpdatedEvent         ] - Thing 'zwave:device:f1150c69:node4' has been updated.
> 2017-08-03 20:30:30.158 [ConfigStatusInfoEvent     ] - ConfigStatusInfo [configStatusMessages=[]]
> 2017-08-03 20:30:30.224 [ConfigStatusInfoEvent     ] - ConfigStatusInfo [configStatusMessages=[]]
> 2017-08-03 20:30:30.227 [ThingUpdatedEvent         ] - Thing 'zwave:device:f1150c69:node4' has been updated.
> 2017-08-03 20:30:30.773 [ThingUpdatedEvent         ] - Thing 'zwave:device:f1150c69:node4' has been updated.
> 2017-08-03 20:30:30.776 [ConfigStatusInfoEvent     ] - ConfigStatusInfo [configStatusMessages=[]]
> 2017-08-03 20:30:30.801 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from ONLINE to UNINITIALIZED
> 2017-08-03 20:30:31.028 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from UNINITIALIZED to UNINITIALIZED (HANDLER_MISSING_ERROR)
> 2017-08-03 20:30:31.189 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from UNINITIALIZED (HANDLER_MISSING_ERROR) to INITIALIZING
> 2017-08-03 20:30:31.199 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
> 2017-08-03 20:30:31.232 [ThingUpdatedEvent         ] - Thing 'zwave:device:f1150c69:node4' has been updated.
> 2017-08-03 20:30:31.253 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to ONLINE
> 2017-08-03 20:30:31.256 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from ONLINE to UNINITIALIZED
> 2017-08-03 20:30:31.358 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from UNINITIALIZED to UNINITIALIZED (HANDLER_MISSING_ERROR)
> 2017-08-03 20:30:31.440 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from UNINITIALIZED (HANDLER_MISSING_ERROR) to INITIALIZING
> 2017-08-03 20:30:31.449 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
> 2017-08-03 20:30:31.457 [hingStatusInfoChangedEvent] - 'zwave:device:f1150c69:node4' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to ONLINE
> 2017-08-03 20:30:31.481 [ThingUpdatedEvent         ] - Thing 'zwave:device:f1150c69:node4' has been updated.

Thank you.

cdjackson, 2017-08-04:
> OH2 Version: Openhan2

What version?

The device is already in the database. Probably you have some other issue or you have an old version.

hansaya, 2017-08-07:
Sorry I should have mentioned the version properly. I have openHAB 2.1.0-1

cdjackson, 2017-08-08:
I would suggest to update to the latest snapshot which should include this device. Alternatively there is another issue other than the database.

hansaya, 2017-08-08:
Thank you, I thought I have the latest snapshot but turns out I'm wrong. I'll have to figure out how to update to the latest snapshot. Closing the issue.

closed 2017-08-08 by hansaya