openhab/org.openhab.binding.zwave #650: Add Alarm Level for Alarm 19 Events to capture userid's to developmental binding

rgerrans, 2017-08-05:
As per:  https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653/1405?u=rgerrans

Sample debug communication logs:
```

2017-08-05 08:29:40.625 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 1E 00 04 00 57 18 98 81 E9 BC 39 BA D2 B8 CE C5 71 6F 28 56 EF C1 05 49 0F 55 C8 CD 04 33 6E 
2017-08-05 08:29:40.628 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Response SENT 6
2017-08-05 08:29:40.630 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=87, callback=0, payload=00 57 18 98 81 E9 BC 39 BA D2 B8 CE C5 71 6F 28 56 EF C1 05 49 0F 55 C8 CD 04 33 
2017-08-05 08:29:40.632 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=87, callback=0, payload=00 57 18 98 81 E9 BC 39 BA D2 B8 CE C5 71 6F 28 56 EF C1 05 49 0F 55 C8 CD 04 33 
2017-08-05 08:29:40.634 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=87, callback=0, payload=00 57 18 98 81 E9 BC 39 BA D2 B8 CE C5 71 6F 28 56 EF C1 05 49 0F 55 C8 CD 04 33 
2017-08-05 08:29:40.634 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2017-08-05 08:29:40.635 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 87: Application Command Request (ALIVE:DONE)
2017-08-05 08:29:40.636 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 87: resetResendCount initComplete=true isDead=false
2017-08-05 08:29:40.636 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 87: Decapsulating COMMAND_CLASS_SECURITY
2017-08-05 08:29:40.638 [DEBUG] [mmandclass.ZWaveSecurityCommandClass] - NODE 87: SECURITY_RXD 71 05 13 03 
2017-08-05 08:29:40.639 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 87: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2017-08-05 08:29:40.639 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 87: Received COMMAND_CLASS_ALARM V1 NOTIFICATION_REPORT
2017-08-05 08:29:40.640 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 87: ALARM report - 19 = 3
2017-08-05 08:29:40.640 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 87: Alarm Type = null (19)
2017-08-05 08:29:40.641 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
2017-08-05 08:29:40.641 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 87: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2017-08-05 08:29:40.642 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 87: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 3
2017-08-05 08:29:40.643 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing ALARM
2017-08-05 08:29:40.644 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 87: Updating channel state zwave:device:54545887:node87:alarm_number to 19 [DecimalType]
2017-08-05 08:29:40.647 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 87: Commands processed 1.
2017-08-05 08:29:40.647 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 87: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@1d38d1.
2017-08-05 08:29:40.648 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0

And from 5iver:

```
2017-08-05 10:42:03.954 [ZWaveNode                 ] - NODE 183: Decapsulating COMMAND_CLASS_SECURITY
2017-08-05 10:42:03.954 [ZWaveSecurityCommandClass ] - NODE 183: SECURITY_RXD 71 05 13 01 00 FF 06 06 01 01 
2017-08-05 10:42:03.954 [ZWaveNode                 ] - NODE 183: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2017-08-05 10:42:03.954 [ZWaveCommandClass         ] - NODE 183: Received COMMAND_CLASS_ALARM V3 NOTIFICATION_REPORT
2017-08-05 10:42:03.955 [ZWaveAlarmCommandClass    ] - NODE 183: NOTIFICATION report - 19 = 1, event=6, status=255, plen=1
2017-08-05 10:42:03.955 [ZWaveAlarmCommandClass    ] - NODE 183: Alarm Type = ACCESS_CONTROL (19)
2017-08-05 10:42:03.955 [ZWaveController           ] - Notifying event listeners: ZWaveAlarmValueEvent
2017-08-05 10:42:03.955 [ZWaveThingHandler         ] - NODE 183: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2017-08-05 10:42:03.955 [ZWaveThingHandler         ] - NODE 183: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 255
2017-08-05 10:42:03.955 [ZWaveAlarmConverter       ] - Alarm converter processing NOTIFICATION
2017-08-05 10:42:03.955 [ZWaveAlarmConverter       ] - Alarm converter NOTIFICATION event is 6, type OnOffType
2017-08-05 10:42:03.955 [ZWaveAlarmConverter       ] - Alarm converter NOTIFICATION event is 6, channel alarm_general is not implemented.
2017-08-05 10:42:03.955 [ZWaveAlarmConverter       ] - Alarm converter processing NOTIFICATION
2017-08-05 10:42:03.955 [ZWaveAlarmConverter       ] - Alarm converter NOTIFICATION event is 6, type OnOffType
2017-08-05 10:42:03.955 [ZWaveThingHandler         ] - NODE 183: Updating channel state zwave:device:07cb40a2:node183:notification_access_control to 6 [DecimalType]
2017-08-05 10:42:03.960 [ZWaveAlarmConverter       ] - Alarm converter processing NOTIFICATION
2017-08-05 10:42:03.960 [ZWaveAlarmConverter       ] - Alarm converter NOTIFICATION event is 6, type DecimalType
2017-08-05 10:42:03.960 [ZWaveThingHandler         ] - NODE 183: Updating channel state zwave:device:07cb40a2:node183:alarm_number to 6 [DecimalType]
2017-08-05 10:42:03.966 [ZWaveTransactionManager   ] - NODE 183: Commands processed 1.

```
```

cdjackson, 2017-08-09:
I will add a new channel called "alarm_raw".  It will be a string type and will have the following json format -:

```
{"type":"ACCESS_CONTROL","event":"6","status":"255"}
```

Hopefully this will be ok?

Feel free to add this channel to the database - the mods are done already for notifications (ie from 5ivers log). It might take a little more thought to get the alarms included as they are less well defined and the format might change slightly in this mode.

rgerrans, 2017-08-09:
I think that should work.

To add it do I just need to reference it like below or do I need to put in the format above in the config field:

[image]

cdjackson, 2017-08-09:
This is fine as it is (i.e. no config) but the channel name needs to be lower case (i.e. alarm_raw).

rgerrans, 2017-08-09:
Oops, total rookie mistake.  It's been added to 914TRL but generated the following error:
`
Channel in endpoint 0 uses an unsupported channel (alarm_raw)!
`

cdjackson, 2017-08-09:
Don’t worry about the error - I need to add it to the database exporter, but I’ll sort this out….

rgerrans, 2017-08-09:
Great, thanks.  Just let me know when there is a compiled version ready to test and I'll take a run at it.

openhab-5iver, 2017-08-09:
Chris, could you include alarm 18 events, so we can also catch codes for locking?

cdjackson, 2017-08-10:
> Chris, could you include alarm 18 events, so we can also catch codes for locking?

It will handle ALL events.

> but can't submit it for review without the exporter.

Don't worry - make the changes and I'll sort the rest.

rgerrans, 2017-08-21:
Chris,

Just wanted to check in to see if you were close to this being ready for testing?

cdjackson, 2017-08-23:
It's all coded for notifications - but not the older version which you (@rgerrans) have.  I'll see if I can add the older class in shortly.

rgerrans, 2017-08-23:
@cdjackson Thanks.  Do you mean the older version of the binding or my locks?

Thanks again!!

Ron

cdjackson, 2017-08-23:
Neither - I meant the older version of the command class… The ALARM/NOTIFICATION command class has many versions. Your locks use an old version of this class an I’ve not implemented this for this function yet.

rgerrans, 2017-08-23:
Thanks for the clarification.

cdjackson, 2017-10-24:
I believe this is all working now so closing...

closed 2017-10-24 by cdjackson