openhab/org.openhab.binding.zwave #654: Please add Danfoss HVR CCM to the database

pravussum, 2017-08-09:
My Danfoss HVR ZWave node is not fully recognised by openhab. I get the usual "Unknown device" error message ("This device has not been fully discovered by the binding"... and so on).

Z-Wave Node 4 (0002:8007:0202:2.7)

It seems the device is not in the database.

The device had already been added to the openhab 1 zwave addon database:
https://github.com/openhab/openhab1-addons/issues/1930

However, I can't find it in the ESH-INF/thing directory now. 

* OH2 Version: openHAB 2.1.0 - release build -
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): binding-zwave - 2.1.0

If the problem is associated with a device, please provide the following -:

* Device name: Danfoss - Heat Recovery Ventilation - Computing and Communications Module
* Device version: 
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/166

cdjackson, 2017-08-09:
Please see information on adding / updating database here -:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide>

You should upload your XML to the database.

pravussum, 2017-08-10:
Ah, OK, I see. Can you give me update access so I can edit the existing entry, please? Username is pravussum as well.

cdjackson, 2017-08-10:
Thanks - that's updated. Any problems, just let me know.

closed 2017-08-16 by cdjackson