openhab/org.openhab.binding.zwave #665: Null Pointer Exception while updating device properties

nikooo777, 2017-08-22:
By simply changing the [location](https://scrn.storni.info/2017-08-22_18-20-42-083374328.png) in the device settings, the binding throws a null pointer exception and the device stops updating until completely removed and re-added. 

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version: 2.2.0-snapshot (forked and compiled locally)
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): Aug 16 20:04 zwave-2.2.0-SNAPSHOT

If the problem is associated with a device, please provide the following -:
The problem is most likely not uniquely associated with the device, however I couldn't test it with other devices just yet so I'll list it

* Device name: PST02-1C - Slim Multi-Sensor (Door/Temperature/Illumination)
* Device version: unsure
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/170
* If the device is not in the database, provide a link to the manual. https://shop.swiss-domotique.ch/en/index.php?controller=attachment&id_attachment=402
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.
```
17:59:21.249 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:442cc5bd:node5' changed from ONLINE to ONLINE: Node initialising: EMPTYNODE
17:59:21.325 [ERROR] [ome.io.rest.core.thing.ThingResource] - Exception during HTTP PUT request for update config at 'things/zwave:device:442cc5bd:node5/config' for thing 'zwave:device:442cc5bd:node5': For input string: "org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@7a867006"
17:59:21.327 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_442cc5bd_serial_ack changed from 144 to 145
17:59:21.331 [INFO ] [smarthome.event.BindingEvent        ] - org.openhab.binding.zwave.event.BindingEvent@79fbfc5e
17:59:21.334 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_442cc5bd_serial_sof changed from 280 to 281
17:59:21.336 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_442cc5bd_serial_sof changed from 281 to 282
17:59:21.337 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_442cc5bd_serial_ack changed from 145 to 146
17:59:21.424 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:442cc5bd:node5' changed from ONLINE: Node initialising: EMPTYNODE to ONLINE: Node initialising: PROTOINFO
17:59:21.472 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:442cc5bd:node5' changed from ONLINE to OFFLINE (BRIDGE_OFFLINE): Controller is offline
17:59:21.483 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:442cc5bd:node5' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to ONLINE
17:59:21.471 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while calling thing updated at ThingHandler 'org.openhab.binding.zwave.handler.ZWaveThingHandler@1f7e6898': java.lang.NullPointerException
java.util.concurrent.ExecutionException: java.lang.NullPointerException
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_131]
	at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_131]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:194)
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:83)
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:67)
	at org.eclipse.smarthome.core.thing.internal.ThingManager.thingUpdated(ThingManager.java:524)
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyTrackers(ThingRegistryImpl.java:209)
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyListenersAboutUpdatedElement(ThingRegistryImpl.java:132)
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyListenersAboutUpdatedElement(ThingRegistryImpl.java:1)
	at org.eclipse.smarthome.core.common.registry.AbstractRegistry.updated(AbstractRegistry.java:183)
	at org.eclipse.smarthome.core.common.registry.AbstractRegistry.updated(AbstractRegistry.java:1)
	at org.eclipse.smarthome.core.common.registry.AbstractProvider.notifyListeners(AbstractProvider.java:57)
	at org.eclipse.smarthome.core.common.registry.AbstractProvider.notifyListenersAboutUpdatedElement(AbstractProvider.java:81)
	at org.eclipse.smarthome.core.common.registry.AbstractManagedProvider.update(AbstractManagedProvider.java:124)
	at org.eclipse.smarthome.io.rest.core.thing.ThingResource.update(ThingResource.java:437)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)[:1.8.0_131]
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)[:1.8.0_131]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)[:1.8.0_131]
	at java.lang.reflect.Method.invoke(Method.java:498)[:1.8.0_131]
	at org.glassfish.jersey.server.model.internal.ResourceMethodInvocationHandlerFactory$1.invoke(ResourceMethodInvocationHandlerFactory.java:81)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher$1.run(AbstractJavaResourceMethodDispatcher.java:144)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.invoke(AbstractJavaResourceMethodDispatcher.java:161)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.JavaResourceMethodDispatcherProvider$ResponseOutInvoker.doDispatch(JavaResourceMethodDispatcherProvider.java:160)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.dispatch(AbstractJavaResourceMethodDispatcher.java:99)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:389)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:347)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:102)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.ServerRuntime$2.run(ServerRuntime.java:326)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.internal.Errors$1.call(Errors.java:271)[155:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors$1.call(Errors.java:267)[155:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors.process(Errors.java:315)[155:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors.process(Errors.java:297)[155:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors.process(Errors.java:267)[155:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:317)[155:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.server.ServerRuntime.process(ServerRuntime.java:305)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.ApplicationHandler.handle(ApplicationHandler.java:1154)[156:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.servlet.WebComponent.serviceImpl(WebComponent.java:473)[153:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
	at org.glassfish.jersey.servlet.WebComponent.service(WebComponent.java:427)[153:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
	at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:388)[153:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
	at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:341)[153:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
[... 18 lines cut ...]
[... cut ...]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_131]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_131]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_131]
	... 1 more
17:59:21.608 [INFO ] [smarthome.event.ThingUpdatedEvent   ] - Thing 'zwave:device:442cc5bd:node5' has been updated.

```

It does look like the exception is being thrown inside the binding, sorry if that's not the case.

BR,
Niko

cdjackson, 2017-08-22:
Please can you provide a debug log showing the problem.

Thanks

nikooo777, 2017-08-31:
Hi, sorry for my lack of input.

Could you please explain me what you mean by debug log? I thought that log was it but I'm apparently mistaken.
BR

cdjackson, 2017-09-04:
The debug log is actually just the standard openhab log file, but with debugging enabled. To enable debugging log on to the console and use the command log:set DEBUG org.openhab.binding.zwave.

friedrichteucher, 2017-09-14:
Hi, ran across the same issue lately with PST02-1A changing device location in HABmin in openhab/zwave 2.1.0.
````
2017-09-12 20:55:30.709 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while calling thing updated at ThingHandler 'org.openhab.binding.zwave.handler.ZWaveThingHandler@116b83c': java.lang.NullPointerException
java.util.concurrent.ExecutionException: java.lang.NullPointerException
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_121]
	at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_121]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:194)
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:83)
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:67)
	at org.eclipse.smarthome.core.thing.internal.ThingManager.thingUpdated(ThingManager.java:524)
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyTrackers(ThingRegistryImpl.java:221)
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyListenersAboutUpdatedElement(ThingRegistryImpl.java:144)
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyListenersAboutUpdatedElement(ThingRegistryImpl.java:1)
	at org.eclipse.smarthome.core.common.registry.AbstractRegistry.updated(AbstractRegistry.java:177)
	at org.eclipse.smarthome.core.common.registry.AbstractProvider.notifyListeners(AbstractProvider.java:57)
	at org.eclipse.smarthome.core.common.registry.AbstractProvider.notifyListenersAboutUpdatedElement(AbstractProvider.java:82)
	at org.eclipse.smarthome.core.common.registry.AbstractManagedProvider.update(AbstractManagedProvider.java:133)
	at org.eclipse.smarthome.io.rest.core.thing.ThingResource.update(ThingResource.java:437)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)[:1.8.0_121]
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)[:1.8.0_121]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)[:1.8.0_121]
	at java.lang.reflect.Method.invoke(Method.java:498)[:1.8.0_121]
	at org.glassfish.jersey.server.model.internal.ResourceMethodInvocationHandlerFactory$1.invoke(ResourceMethodInvocationHandlerFactory.java:81)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher$1.run(AbstractJavaResourceMethodDispatcher.java:144)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.invoke(AbstractJavaResourceMethodDispatcher.java:161)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.JavaResourceMethodDispatcherProvider$ResponseOutInvoker.doDispatch(JavaResourceMethodDispatcherProvider.java:160)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.dispatch(AbstractJavaResourceMethodDispatcher.java:99)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:389)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:347)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:102)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.ServerRuntime$2.run(ServerRuntime.java:326)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.internal.Errors$1.call(Errors.java:271)[157:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors$1.call(Errors.java:267)[157:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors.process(Errors.java:315)[157:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors.process(Errors.java:297)[157:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.internal.Errors.process(Errors.java:267)[157:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:317)[157:org.glassfish.jersey.core.jersey-common:2.22.2]
	at org.glassfish.jersey.server.ServerRuntime.process(ServerRuntime.java:305)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.server.ApplicationHandler.handle(ApplicationHandler.java:1154)[158:org.glassfish.jersey.core.jersey-server:2.22.2]
	at org.glassfish.jersey.servlet.WebComponent.serviceImpl(WebComponent.java:473)[155:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
	at org.glassfish.jersey.servlet.WebComponent.service(WebComponent.java:427)[155:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
	at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:388)[155:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
	at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:341)[155:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
	at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:228)[155:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
[... cut ...]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$7.call(ThingManager.java:1)[105:org.eclipse.smarthome.core.thing:0.9.0.b5]
	at org.eclipse.smarthome.core.common.SafeMethodCaller$CallableWrapper.call(SafeMethodCaller.java:181)[98:org.eclipse.smarthome.core:0.9.0.b5]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_121]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_121]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_121]
	... 1 more
````
So far I have not been able to re-add the device to reproduce.

nikooo777, 2017-09-14:
you'll have to:
1. hard reset the controller
2. reset the device
3. remove the controller from OH
4. restart OH
5. add the controller again
6. include/add the device
7. DO NOT EDIT the device otherwise everything breaks again
8. if you really want to edit the device then use habadmin to do it

I'll come back with a debug log when i have time for it, unfortunately with a strict schedule I haven't been able to so far.

cdjackson, 2017-09-14:
> you'll have to:
>
> hard reset the controller
> reset the device
> remove the controller from OH
> restart OH
> add the controller again
> include/add the device
> DO NOT EDIT the device otherwise everything breaks again
> if you really want to edit the device then use habadmin to do it
>
>
> I'll come back with a debug log when i have time for it, unfortunately with a strict schedule I haven't been able to so far.

I'm not sure what this error is,, but there should be no reason to hard reset the controller or the device.

friedrichteucher, 2017-09-14:
Was able to reinclude the device again - had to reset device and exclude first. It is now shown up as node 18 - cf. openhab.log to check inclusion and several changes of 'location' property in HABmin before and after items where linked: 
[openhab.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1303692/openhab.txt)
However, have not been able to reproduce:-(

One thing I noticed is that parameter 7 is shown twice. 
[image]
Will update if NPE happens again during further setup.

cdjackson, 2017-12-30:
> Will update if NPE happens again during further setup.

Assuming this hasn't happened again so closing. Please reopen if it still happening.

closed 2017-12-30 by cdjackson