openhab/org.openhab.binding.zwave #670: Regeneration of userdata/zwave/xxx.xml-files is incomplete

brstein, 2017-08-28:
Setup:
* OpenHAB 2.1.0 release build
* ZWave-Binding devel-snapshot 2.1.0.201708272249
* Z-Stick Gen 5
* Multisensor 6 (secure inclusion)

I included the Multisensor 6 with security using HABmin (nothing textually defined). The multisensor's thing-properties in HABmin confirm the security. When I use the Z-Stick with a textually defined configuration (things and items defined textually), HABmin states that no security is used. If I also transfer the multisensors xml in userdata/zwave to the textually defined instance everything works fine and the security is confirmed.

As you mentioned in [the forum ](https://community.openhab.org/t/openhab2-zwave-textual-configuration-a-discussion/33424/2) a missing file in userdata/zwave should be regenerated. The only effect should be a slower start.

However the regenerated file is much shorter. For example the secureCommandClasses-Section is empty and several other settings (s. attachment) are not filled.
Further more I observed, that parameters like temperature calibration are not queried from the device and will not appear in the XML-file.

Regards, brstein

Attachments: XML generated at inclusion and XML regenerated when missing:

[network_cb4f1afc__node_2.xml_inclusion.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1257270/network_cb4f1afc__node_2.xml_inclusion.txt)
[network_cb4f1afc__node_2.xml_regenerated.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1257271/network_cb4f1afc__node_2.xml_regenerated.txt)

cdjackson, 2017-08-28:
Try to delete the XML again to see if it works. When the binding restarts without the XML it performs a handshake with the device to see if it responds to security - if it doesn’t get the response it needs, then it defaults to security off.

Also make sure that the security key is also the same you originally used - otherwise it will never work.

cdjackson, 2017-08-28:
I don't think there's a bug here so let's discuss on the forum so others can participate.

closed 2017-08-28 by cdjackson

brstein, 2017-08-28:
Security key is the same for both instances. 
The problem can be reproduced on the textually defined instance only.
If I delete the xml on the non-textually defined instance it seems to be rebuild properly.
The problem arises only in the textually defined instance - ceteris paribus.

Further note: There should not be any configuration problem (security key, things- and items-file) because it works immediately when the xml created at inclusion is transfered to the textually defined instance.

cdjackson, 2017-08-28:
Ok, then I’ll need more information. If it’s only configuration, then it’s very likely a problem with the configuration itself. The binding knows nothing at all about how you have configured the device - i.e. if it is done through text files or through the UI - it’s exactly the same as far as the binding is concerned so probably you have something misconfigured.

brstein, 2017-08-28:
Thanks for your help! 

Here come the files (I added a .txt to get them uploaded):

* [zwave.things.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1257524/zwave.things.txt)
* [zwave.items.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1257522/zwave.items.txt)
* [default.sitemap.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1257543/default.sitemap.txt)

As mentioned before:
* It works well when copying the XML-file for node 2 (which is the multisensor) from the non-textually defined instance. I get the proper sensor readings in the BasicUI, while the security is confirmed in HabMIN.
* Regeneration of the XML-files fails only in the scripted environment.

cdjackson, 2017-08-28:
I’m not really sure I can do much with this - there must be something wrong (i.e. different) with the configuration. As I said, the binding doesn’t know anything about text files or UI configuration etc. I know that the XML files are recreated fine normally - even for secure devices. This has been tested many times by many people so here the difference (if I understand correctly) is just that you are configuring through the text files?

Can you compare or provide your configuration between the two systems? Are you 100% sure the key is the same? What do the logs show is happening? Do other (non secure) devices work properly (if so, again I would think it’s the security key that’s different as it’s about the only thing that can be set).

brstein, 2017-08-28:
I am totally sure that the key is the same. I compared it in HABmin (non-textual-config) and the things-file (s. above). Further more I get the correct sensor-readings when transferring the xml-file.

Could the textual configuration be a problem when querying the node, because other configuration options are also limited when using textual configuration?

I added the log file (debug-mode), which is produced during the creation of the xml-file. Maybe it helps: [openhab.log.zip](https://github.com/openhab/org.openhab.binding.zwave/files/1257706/openhab.log.zip)

cdjackson, 2017-08-28:
> Could the textual configuration be a problem when querying the node, because other configuration options are also limited when using textual configuration?

Sorry - can you explain what you mean by this? As above, the binding doesn't know where the configuration came from (UI or text files) so I don't think it's an issue, but please can you explain this better so I can understand your point.

cdjackson, 2017-08-28:
I suspect your problem is the following configuration -:

```
security_inclusionmode="1"
```

This should be set as an integer - not a string.

cdjackson, 2017-08-28:
Note that the above issue is also relevant to other configuration options in your config...

brstein, 2017-08-29:
You are right! The problem was

     security_inclusionmode="1"

which should be provided as integer. If I define it as =1 the regeneration works as expected.

I was mislead by the fact, that the GUI of HABmin as well as the GUI of PaperUI showed up: Secure Inclusion Mode: All Devices. **Therefore my question: Are the integers provided as string values casted in different ways to integer in the program?** (Which could be the explanation for this observation)

Further I discovered, when I use the non-textual approach, the regeneration of the XMLs also fails if I set the "Secure Inclusion Mode" to locks only on the GUI. I was expecting that this parameter is only necessary for inclusion but not for further operation after inclusion. Is this behaviour as expected?

Thank your kind help.

cdjackson, 2017-08-29:
No. Integers provided in other formats aren't cast/converted. The UI does a conversion between the text strings and the integers. 

If you change the value of the inclusion type, then yes, it's expected that it won't work if you delete the xml since it won't attempt to perform the secure handshake on restart. 

So everything  is explained I think by the configuration.