openhab/org.openhab.binding.zwave #671: S0 lock mode lock control is invalid

Ynoy, 2017-08-29:
The following is the generated log:
10:26:22.952 [WARN ] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: No command converter set for command zwave:device:7b321dc8:node4:switch_binary type OnOffType
10:26:22.958 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'zwave_device_7b321dc8_node4_switch_binary' received command OFF
10:26:22.970 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_device_7b321dc8_node4_switch_binary changed from ON to OFF
10:26:23.455 [WARN ] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: No command converter set for command zwave:device:7b321dc8:node4:lock_door type OnOffType
10:26:23.459 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'zwave_device_7b321dc8_node4_lock_door' received command OFF
10:26:23.469 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_device_7b321dc8_node4_lock_door changed from ON to OFF
10:26:26.839 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_7b321dc8_serial_sof changed from 511 to 512
10:26:35.970 [WARN ] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: No command converter set for command zwave:device:7b321dc8:node4:switch_binary type OnOffType
10:26:35.972 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'zwave_device_7b321dc8_node4_switch_binary' received command ON
10:26:35.981 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_device_7b321dc8_node4_switch_binary changed from OFF to ON

I used the binding.zwave.jar that is the link
https://github.com/cdjackson/org.openhab.binding.zwave/tree/development
Request support！

cdjackson, 2017-08-29:
Please provide more information on your system. What devices are used for starters is needed. A debug log would be useful showing the initialataion of the device.

cdjackson, 2017-12-09:
No feedback provided so closing. Feel free to reopen once the requested information is provided.

closed 2017-12-09 by cdjackson