openhab/org.openhab.binding.zwave #691: update database PAN08_1a

x-dragos, 2017-09-22:
Hi,

i have updated this rollershuter controller to present blinds_cotrol channel instead of switch_dimmer which doesnt support the invert_control and invert_percentage properties, but i dont know how to use github very well .. so i am unsure of how i should propose the update.

i have tested this on my local snapshot install by editing the JAR and reinstalling binding and it works well, aside from the fact that i was able to save the invert_percent only with paperUI and not with Habmin (reason unknown)

attached XML for review. hope this helps someone because it took me 2 days to understand all this and set it up :)

[philio_pan081a_0_0.zip](https://github.com/openhab/org.openhab.binding.zwave/files/1324570/philio_pan081a_0_0.zip)

cdjackson, 2017-09-22:
This will be updated through the database directly. I'll do a review and update tonight probably (otherwise tomorrow).

closed 2017-09-22 by cdjackson

x-dragos, 2017-09-22:
i am sure this sounds stupid ... but i fail to understand how it gets updated. 
i found a temp folder where it probably downloaded a snapshot zwave binding .. .and i edited that one. 
when you update the database (as u said) how will i be able to pull the "official" update ?

(sorry for the noobish version control experience that i have)

cdjackson, 2017-09-22:
The binding will be updated with the new database - you will then need to update to the latest snapshot. If you're on the snapshot runtime, then simply uninstall the binding in PaperUI, wait 30 seconds, then install it again.

If you are using the release version, it's more complex - please discuss on the forum.