openhab/org.openhab.binding.zwave #698: Please add support for GoControl GC-TBZ48

justinpitts, 2017-09-26:
I have included two of these thermostats in my zwave network, but both show as unknown devices with populated mfg and device id.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/622

cdjackson, 2017-09-26:
As you have shown, the device is already included. If it's not showing up, then there's probably another problem. Please discuss on the forum.

closed 2017-09-26 by cdjackson

justinpitts, 2017-09-26:
Your in-product documentation is therefore out-of-date, because it is directing me to "raise an issue."

    The device is not in the database. If the device attributes show that this device has a valid manufacturer ID, device ID and type, then this is likely the case (eg. you see a label like "Z-Wave node 1 (0082:6015:020D::2.0)"). Even if the device appears to be in the database, some manufacturers use multiple sets of references for different regions or versions, and your device references may not be in the database. In either case, the database must be updated and you should raise an issue to get this addressed.