openhab/org.openhab.binding.zwave #699: NullPointerException on synchronized monitor in ZWaveThingHandler

henriksvensson, 2017-09-26:
I'm running OpenHAB on a Raspberry Pi 3 and ran into a NullPointerException in the ZWaveThingHandler class. See log below:

```
2017-09-25 09:58:23.732 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while disposing handler of thing 'zwave:device:bd7c3a44:node24': ja
va.lang.NullPointerException
java.util.concurrent.ExecutionException: java.lang.NullPointerException
        at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_121]
        at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_121]
        at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:194)
        at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:83)
        at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:67)
        at org.eclipse.smarthome.core.thing.internal.ThingManager.doDisposeHandler(ThingManager.java:831)
        at org.eclipse.smarthome.core.thing.internal.ThingManager.disposeHandler(ThingManager.java:819)
        at org.eclipse.smarthome.core.thing.internal.ThingManager.unregisterAndDisposeHandler(ThingManager.java:865)
        at org.eclipse.smarthome.core.thing.internal.ThingManager.unregisterAndDisposeChildHandlers(ThingManager.java:855)
        at org.eclipse.smarthome.core.thing.internal.ThingManager.unregisterAndDisposeHandler(ThingManager.java:863)
        at org.eclipse.smarthome.core.thing.internal.ThingManager.removeThingHandlerFactory(ThingManager.java:1044)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)[:1.8.0_121]
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)[:1.8.0_121]
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)[:1.8.0_121]
        at java.lang.reflect.Method.invoke(Method.java:498)[:1.8.0_121]
        at org.apache.felix.scr.impl.inject.BaseMethod.invokeMethod(BaseMethod.java:224)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.inject.BaseMethod.access$500(BaseMethod.java:39)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.inject.BaseMethod$Resolved.invoke(BaseMethod.java:617)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.inject.BaseMethod$NotResolved.invoke(BaseMethod.java:576)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.inject.BaseMethod.invoke(BaseMethod.java:501)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.inject.BindMethod.invoke(BindMethod.java:655)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.DependencyManager.invokeUnbindMethod(DependencyManager.java:1837)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.SingleComponentManager.invokeUnbindMethod(SingleComponentManager.java:394)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.DependencyManager$MultipleDynamicCustomizer.removedService(DependencyManager.java:375)[31:org.apache.felix.scr:2.0
.6]
        at org.apache.felix.scr.impl.manager.DependencyManager$MultipleDynamicCustomizer.removedService(DependencyManager.java:291)[31:org.apache.felix.scr:2.0
.6]
        at org.apache.felix.scr.impl.manager.ServiceTracker$Tracked.customizerRemoved(ServiceTracker.java:1241)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.ServiceTracker$Tracked.customizerRemoved(ServiceTracker.java:1136)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.ServiceTracker$AbstractTracked.untrack(ServiceTracker.java:996)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.ServiceTracker$Tracked.serviceChanged(ServiceTracker.java:1175)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.BundleComponentActivator$ListenerInfo.serviceChanged(BundleComponentActivator.java:127)[31:org.apache.felix.scr:2.0.6]
        at org.eclipse.osgi.internal.serviceregistry.FilteredServiceListener.serviceChanged(FilteredServiceListener.java:109)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.framework.BundleContextImpl.dispatchEvent(BundleContextImpl.java:914)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.serviceregistry.ServiceRegistry.publishServiceEventPrivileged(ServiceRegistry.java:862)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.serviceregistry.ServiceRegistry.publishServiceEvent(ServiceRegistry.java:801)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.serviceregistry.ServiceRegistrationImpl.unregister(ServiceRegistrationImpl.java:222)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.apache.felix.scr.impl.manager.AbstractComponentManager$3.unregister(AbstractComponentManager.java:908)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.AbstractComponentManager$3.unregister(AbstractComponentManager.java:873)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.RegistrationManager.changeRegistration(RegistrationManager.java:139)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.AbstractComponentManager.unregisterService(AbstractComponentManager.java:950)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.AbstractComponentManager.doDeactivate(AbstractComponentManager.java:805)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.AbstractComponentManager.deactivateInternal(AbstractComponentManager.java:787)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.AbstractComponentManager.dispose(AbstractComponentManager.java:579)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.manager.ConfigurableComponentHolder.disposeComponents(ConfigurableComponentHolder.java:706)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.BundleComponentActivator.dispose(BundleComponentActivator.java:523)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.Activator.disposeComponents(Activator.java:452)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.Activator.access$300(Activator.java:54)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.scr.impl.Activator$ScrExtension.destroy(Activator.java:306)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.utils.extender.AbstractExtender$2.run(AbstractExtender.java:290)[31:org.apache.felix.scr:2.0.6]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_121]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_121]
        at org.apache.felix.utils.extender.AbstractExtender.destroyExtension(AbstractExtender.java:312)[31:org.apache.felix.scr:2.0.6]
        at org.apache.felix.utils.extender.AbstractExtender.bundleChanged(AbstractExtender.java:186)[31:org.apache.felix.scr:2.0.6]
        at org.eclipse.osgi.internal.framework.BundleContextImpl.dispatchEvent(BundleContextImpl.java:902)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEventPrivileged(EquinoxEventPublisher.java:165)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:75)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:67)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
        at org.eclipse.osgi.internal.framework.EquinoxContainerAdaptor.publishModuleEvent(EquinoxContainerAdaptor.java:102)[org.eclipse.osgi-3.10.101.v20150820-1432.jar:]
[... cut ...]

It assumes that `pollingJob` in `synchronized (pollingJob)` statements is never null. The next line, inside of the synchronized code block, checks if `pollingJob` is null, but it never will be. I believe that the synchronized statement should use another object as monitor (maybe `this`?). 

There are three such occurrences in the file and I cannot find any other similar issues in the other classes of the binding.

* OH2 Version: 2.1.0-1
* Binding Version: development branch, 2.1.0.201709251646

cdjackson, 2017-09-26:
Thanks. This was (clearly!) put together a bit quickly - I’ll update.

cdjackson, 2017-10-05:
This should now be resolved

closed 2017-10-05 by cdjackson