openhab/org.openhab.binding.zwave #710: Create Generic Channel for "Technical Status" of a Zwave device

shorty707, 2017-10-14:
Some of this information is available in Habmin but it would be nice to have them as a channel and write these values to an item for sitemap and rule integration.

-> Last Wakeup as seen in Habmin
-> Last “Communication” / "Seen Online"
-> Battery Low (the log somtimes shows an automatic alarm message: NODE XX: BATTERY LOW!) 
-> Status (ONLINE/OFFLINE) as shown in Habmin

Let me know if you think this makes sense. Cheers

sihui62, 2017-10-14:
> Status (ONLINE/OFFLINE) as shown in Habmin

This is already provided through ESH functionality, example (copied from the forum):

```
rule "zwave node xx"
	when 
		Thing "zwave:device:yxyxyxyxyx:nodexx" changed
	then
		var status = ThingAction.getThingStatusInfo("zwave:device:15ca6a108b9:node26").getStatus()
		if (status.toString() == 'OFFLINE')
		{
			logInfo("test","node xx offine")		
		}
		else {
			logInfo("test","node xx online")	
		}
end
```

sihui62, 2017-10-14:
For reference: this was already discussed in the community forum:

https://community.openhab.org/t/zwave-technical-status-of-devices/35094

cdjackson, 2017-10-14:
As mentioned, this was discussed in the forum. Thing status is not a ZWave issue - it's a framework issue and should be available.

closed 2017-10-14 by cdjackson

mhilbush, 2017-10-14:
The one thing I see that might be useful is a channel for last the wakeup time.  You could put the items linked to the wakeup channels in a group, then use a rule to monitor for any item for which a wake-up hasn't occurred in NN minutes.  This would be useful to detect battery-powered devices for which the battery reports don't work so well, such as my Aeon ZW100 multisensor.  😉

cdjackson, 2017-10-14:
I don’t disagree, but in my mind at least, checking that devices are still communicating seems like it should be a standard feature of the framework rather than something that every binding does

shorty707, 2017-10-15:
As you linked in the forum there is something for ONLINE / OFFLINE state. I will check that.

For "Wakeup" .. isn't this Zwave specific?
I mean I would really like something reliable to get notified actively when a "node" / "thing" seems dead

cdjackson, 2017-10-15:
Yes, wakeup concept is specific to ZWave, but my point is that what you are looking for is something to tell if the thing is dead and this is not zwave specific. We could mark a device as offline if it doesn’t wake up (we already have a feature to set it dead if this happens - I could take it offline as well).

shorty707, 2017-10-15:
no thats not good.
there are devices like the everspring smoke detector that by design never wakeup

sihui62, 2017-10-15:
> like the everspring smoke detector

That device is a piece of crap ...
But the functionality you want to have is already available with the thing on-/offline status code from above:

Last wakeup for my device:
[image]

And the on-/offline log for that event shows:

`15-Okt-2017 10:53:35.317 [INFO ] [org.eclipse.smarthome.model.script.EXTRA          ] - ON_OFFLINE: node 26 offline`

which matches exactly that point of time.

cdjackson, 2017-10-15:
Now I’m confused. You asked for a “last wakeup time” - now you say that this won’t work since the device never wakes up.

I’m not really sure what you are looking for. I thought that it is a way to detect if the device is offline? If so, the mechanism I said above will work so long as the device sends data at some regular interval. If it doesn’t, then there’s no way to tell at all.

shorty707, 2017-10-15:
sorry.  the last post was only for this one specific device why I would not think its good to have the binding to set this device to DEAD or OFFLINE as you described above. There might be other devices that never wake up. and yes the everspring smoke sensor is crap just for this fact it never wakes up.

for any other device I would like to have an overview and notification when the device is OFFLINE or did not wakeup (or send anything) > xx hours / days.

I will check for the options you and shiui mentioned. maybe thats already it.

cdjackson, 2017-10-15:
So this device NEVER sends any data unless there’s a fire? Wow - gotta say that’s really bad news if so.  Note that this feature is ALREADY in the binding - devices that don’t send data for (I think) 2 x the wakeup period are marked DEAD - that is there now.

> for any other device I would like to have an overview and notification when the device is OFFLINE or did not wakeup (or send anything) > xx hours / days.
> 

As mentioned, this is already possible. If the thing is offline then the framework provides a way to tell.

shorty707, 2017-10-15:
correct. 
fire
low battery (no regular battery updates!)
and when physical button is pushed
are the only communication

this means many many months with zero communication. this device really sucks.
when I restart OH I have to push the button a couple times...otherwise it would never go through initalization.

cdjackson, 2017-10-15:
Yep - that sounds like bad news for something that you really want to know is working.

In this case though, there’s no way to do what you want to do.