openhab/org.openhab.binding.zwave #715: Eurotronic Spirit is missing in the database

cFire112, 2017-10-18:
Hello,
can anyone pleas add the Eurotronic Spirit Zwave Plus thermostat to the Database?
The Link to the cd-jackson database is: [LINK](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/710?layout=openhab2)

THANK YOU

cFire112, 2017-10-18:
So I have the problem that my Zwave Binding show "unknown device"
BUT I installed the OpenHAB system just 2 hours ago... so how can I fix the problem? The Binding should be the latest version...

cdjackson, 2017-10-18:
Please discuss on the forum as there will be more people to help you there. 

Make sure you wake the device up so it is discovered and make sure you are running g a recent snapshot version(i.e. Not the release version)

closed 2017-10-18 by cdjackson

Burrito-Bazooka, 2017-11-22:
@cdjackson Thanks for replying to this issue. Can you please clarify: what do we need a snapshot version of (as opposed to release version)? The binding plugin, or OpenHab entirely?

thanks

cdjackson, 2017-11-22:
A recent snapshot of the binding is required. Probably you can use the 2.1 release version of OH but I'm not 100% sure that there have been no breaking changes (I don't think so though).