openhab/org.openhab.binding.zwave #716: callback is missing

maxlein, 2017-10-20:
Hi, 
came here from openhab2-addons issue tracker.

Have a strane exception here, and removing/adding in cabman doesn't help...

Exception with ZWave Plus device PST02A 
Openhab version 2.1.0:

```java
017-10-01 14:15:41.711 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-10-01 14:15:41.714 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0C 00 04 00 0A 06 31 05 01 0A 00 4B 8F 
2017-10-01 14:15:41.717 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0C 00 04 00 0A 06 31 05 01 0A 00 4B 8F 
2017-10-01 14:15:41.721 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 0A 06 31 05 01 0A 00 4B 
2017-10-01 14:15:41.723 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 10: Application Command Request (ALIVE:DONE)
2017-10-01 14:15:41.725 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 10: Starting initialisation from DONE
2017-10-01 14:15:41.727 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@1d4b448 already registered
2017-10-01 14:15:41.738 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 10: Incoming command class SENSOR_MULTILEVEL
2017-10-01 14:15:41.740 [DEBUG] [ss.ZWaveMultiLevelSensorCommandClass] - NODE 10: Received COMMAND_CLASS_SENSOR_MULTILEVEL command V5
2017-10-01 14:15:41.742 [DEBUG] [ss.ZWaveMultiLevelSensorCommandClass] - NODE 10: Sensor Multi Level REPORT received
2017-10-01 14:15:41.757 [DEBUG] [ss.ZWaveMultiLevelSensorCommandClass] - NODE 10: Sensor Type = Temperature(1), Scale = 1
2017-10-01 14:15:41.759 [DEBUG] [ss.ZWaveMultiLevelSensorCommandClass] - NODE 10: Sensor Value = 75
2017-10-01 14:15:41.760 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveMultiLevelSensorValueEvent
2017-10-01 14:15:41.761 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 10: Got an event from Z-Wave network: ZWaveMultiLevelSensorValueEvent
2017-10-01 14:15:41.763 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 10: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_MULTILEVEL, value = 75
2017-10-01 14:15:41.764 [DEBUG] [erter.ZWaveMultiLevelSensorConverter] - NODE 10: Sensor is reporting scale 1, requiring conversion to 0. Value is now 75.
2017-10-01 14:15:41.766 [DEBUG] [converter.ZWaveCommandClassConverter] - Converted temperature from 75F to 23.9C
2017-10-01 14:15:41.767 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 10: Updating channel state zwave:device:15a670b08a0:node10:sensor_temperature to 23.9 [DecimalType]
2017-10-01 14:15:41.768 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.IllegalStateException: Could not update state, because callback is missing
	at org.eclipse.smarthome.core.thing.binding.BaseThingHandler.updateState(BaseThingHandler.java:267)[105:org.eclipse.smarthome.core.thing:0.9.0.b5]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.ZWaveIncomingEvent(ZWaveThingHandler.java:1104)[189:org.openhab.binding.zwave:2.1.0]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[189:org.openhab.binding.zwave:2.1.0]
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveMultiLevelSensorCommandClass.handleApplicationCommandRequest(ZWaveMultiLevelSensorCommandClass.java:150)[189:org.openhab.binding.zwave:2.1.0]
	at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:119)[189:org.openhab.binding.zwave:2.1.0]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[189:org.openhab.binding.zwave:2.1.0]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[189:org.openhab.binding.zwave:2.1.0]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[189:org.openhab.binding.zwave:2.1.0]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[189:org.openhab.binding.zwave:2.1.0]
```

cdjackson, 2017-10-20:
Please provide full debug logs including startup.

maxlein, 2017-10-20:
There you go.
Beginning of the log is shortly before restarting openhab.
At 18:55:38.767 the exception occurs. Before the node was unknown, until i played around till openhab got the configuration messages.

[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/1402713/openhab.log)

cdjackson, 2017-10-20:
Thanks. The real error happens a lot earlier than this.

What system are you using? Is it a slow computer (RPi) or something faster?

maxlein, 2017-10-20:
RPi 3

maxlein, 2017-10-22:
Any more information on this? Can I fix this?

cdjackson, 2017-10-22:
Sorry - I’m on holiday. If you want to fix it, then please do - it would be appreciated if you can track it down.

maxlein, 2017-10-22:
Oh I see. Thought you know where the error is as you said it happens a lot earlier..

cdjackson, 2017-10-22:
That comment was just based on your log. I’ve not had the chance to look much at the moment - I’ll take a better look later in the week when I’m home.

timmo, 2017-12-17:
@cdjackson Hey Chris, same exception on my RPI with a fibaro motion sensor. Any idea?

cdjackson, 2018-09-08:
Closing as this is very old. If it still exists, then please provide updated logs.

closed 2018-09-08 by cdjackson