openhab/org.openhab.binding.zwave #720: Update Binding to support all Features of Eurotronic Spirit

cFire112, 2017-10-23:
Please Update the Binding that the features to use an external temperature sensor or to use the radiator as an setting-drive device only are supported.

https://community.openhab.org/t/zwave-eurotronic-spirit-with-external-temperatur-dont-work/35330

chris922, 2017-11-14:
Hi @cdjackson , do you plan to implement this in the next time? Unfortunately we bought this device as it has this external temperature support and now we noticed that we can't use this functionality with our current setup.

I really don't have any clue how complex it is to implement something like this, but as I also want to take a deeper look at those internal z-wave stuff this is maybe a good reason for me to do this.

Would it be better to have an own channel or use the existing `sensor_temperature` channel and just changing the `Number` item will send the report? (this was indeed what I tried before I read that sending those reports is not yet implemented)

I don't know if I will find some time for this and also if I am able to understand enough from the z-wave internals to implement such a thing.. but I like challenges and even if you are faster I think I learned some stuff and can then take a look at the way you added such a support.

Best regards,
Christian

cdjackson, 2017-11-15:
If there is sufficient interest in this (and it seems there are a few people) then I will look at this in the coming few weeks. It requires some thought, and I'll need to purchase the device for testing so give me a few weeks for this please.

chris922, 2017-11-20:
I went though the code, analyzed the multilevel sensor report messages and tried to send a message with the same structure.. but as far as I can see up to now only `SendData` messages are supported to be sent to other nodes. Those reports have the message class `ApplicationCommandHandler` which could only be read.

So if my assumption is correct that sending reports requires this message class there are some deeper changes required in the way messages will be sent.

.. maybe those information are helpful, maybe you already know .. :)

cdjackson, 2017-11-20:
As stated, I will look at this shortly. I’ve ordered up a device and it should be here in the next day or so. I’m well aware of how to add this and it does not require large changes like you suggest. 

If anyone fancies donating to support this to help cover my costs of purchasing the device for test, it would be appreciated.

See the donate button on the bottom of http://www.cd-jackson.com/index.php/about <http://www.cd-jackson.com/index.php/about>

cFire112, 2017-12-03:
Thank you Chris!
I've sent you some money to support you to cover your costs :D 
Hope all functions will be supportet soon

cdjackson, 2017-12-03:
Thanks. I have received the device from Germany (as they don’t appear to be available at the UK supplier I normally use), but unfortunately it doesn’t work :( (out of the hundreds of devices I have here, it’s the first one that won’t even include!).  I will try and make the changes without it as it will likely have to be returned.

I’ll see if I can get something for testing in the next few days though.

cdjackson, 2017-12-09:
The new channel type will be ```sensor_report``` - I've added this to the spirit. I'll do a dev binding update tonight that has code to support this - since I can't test this at the moment I can only check it works with static tests.

I suspect if it doesn't work it will have to wait until I get my replacement device.

chris922, 2017-12-09:
Just give the link to the JAR or the name of the branch so that I can check this with my device. If there is an issue maybe the debug logs are already helpful and if not we will wait until you get your replacement device.

Really miserable that you catched one of the few not working devices.. :(

cdjackson, 2017-12-09:
Unfortunately if it doesn’t work, then probably the logs will not help since it will be down to compatibility with the device and it’s unlikely (although not impossible) that it will provide and response if the format is now what it expects.

I tried emailing Eurotronic but they haven’t yet responded.

The update will be in the development binding - see https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653 <https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653>

grumpyquant, 2018-03-20:
Hi - is there an update on this issue?  I too would be interested in seeing a fix for this.  Thanks.

cdjackson, 2018-03-20:
As above, the feature was implemented already and I've been awaiting feedback for the past few months!

Schluesselmeister, 2018-03-27:
Is the warning "Cannot find channel type: zwave:sensor_report" related to this feature?
I am using the binding dated 18.3.

cdjackson, 2018-03-27:
I've only had a quick look, but I don't think there is a version of the binding dated 18.3. The current version is 17.3 and I did an update last night (so 26.3).

Schluesselmeister, 2018-03-27:
I took the one located here: 
https://openhab.ci.cloudbees.com/job/openHAB2-Bundles/lastStableBuild/

Is it not the same?

cdjackson, 2018-03-27:
This is not the development version and does not have this feature - please see the link above.

Schluesselmeister, 2018-03-27:
Downloaded the right version, removed all nodes and added them back again.
The warning message is still shown.
I am trying to set the temperature of the thermostate, but it is not working as expected (at the moment). But I am still investigating.

cdjackson, 2018-03-27:
It does appear the definition is missing - I'll add it.

Schluesselmeister, 2018-03-27:
OK, waiting for the new version before I continue testing. At the moment it seems that the given temperature (writing to the thermostat temperatur) is not used. The room temperatur is too high.
Setpoint: 20.5°C, actual room temperatur 22°C, value written to the item ("current temperatur"): 25°C
BTW, is the channel "dimmer" used for the valve position?

Schluesselmeister, 2018-03-27:
Strange, now I try to set the configuration parameter 8 to "0". But I cannot as the field only allows to set the external sensor usage.
How can I change the value back to 0 to use the internal temperature sensor?

grumpyquant, 2018-04-05:
I'd like to test this feature now, but I'm a noob - how do I install the development version of the zwave binding?  I assume there's a JAR to download somewhere...?

Thanks - and apols if there is documentation covering this I've missed.

openhab-5iver, 2018-04-05:
Follow the forum link that @cdjackson posted.

cdjackson, 2018-09-08:
Closing as this is very old and should be fixed in the latest development version.

closed 2018-09-08 by cdjackson