openhab/org.openhab.binding.zwave #726: Troubles with FGMS0001 reporting

robkamp, 2017-10-29:
OH2 Version: 2.2.0.201710280843
Zwave Binding:  2.2.0.201710280843

* Device name: Fibargroup FGMS0001
* Device version: 3.2 

* Link to http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/328
* An XML is not being generated for the device 

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.

cdjackson, 2017-10-29:
> Provide a full debug level log of the problem.

??

You might be better off posting a request on the forum, but you're going to need to provide more information than this I'm afraid.

robkamp, 2017-10-29:
Do you want me to post in the OH2 formum. I will do that. I was trying to inly get the zwave binding debug log, but I forgot how to. :-)

cdjackson, 2017-10-29:
It’s probably best to post on the community forum, yes. The reason is that (probably) this is something specific to what you are doing rather than a problem with the binding and there will be a LOT more people to help you there. If we find there is a bug in the binding, then we can open an issue to resolve it.

Don’t worry too much about getting just the zwave logs - the log viewer can easily filter out the “good stuff”.

One thing to look at - have you woken the device up so that it actually initialises? This is probably the most common issue with battery devices.

cdjackson, 2017-11-01:
Discussion moved to forum

closed 2017-11-01 by cdjackson