openhab/org.openhab.binding.zwave #728: COMMAND_CLASS_SIMPLE_AV_CONTROL

Ynoy, 2017-10-31:
@cdjackson @quadule @kabili207 @maxberger @asciiphil 
Hi，Chris
I have an infrared converter device, I found in the process of matching, ZWave Binding did not improve COMMAND_CLASS_SIMPLE_AV_CONTROL. I tried to improve this class myself and then found that adding the device after the generated xml file did not generate COMMAND_CLASS_SIMPLE_AV_CONTROL. Then I directly with a command to control the device, the following situation:

18: 27: 40.961 [INFO] [ommandClassTransactionPayloadBuilder] - commandClass Key 148
18: 27: 40.962 [INFO] [ommandClassTransactionPayloadBuilder] - At build null

I use the method is: new ZWaveCommandClassTransactionPayloadBuilder (getNode (). GetNodeId (), getCommandClass (), SIMPLE_AV_CONTROL_SET) .withPayload (outputData.toByteArray ()). WithPriority (TransactionPriority.Set) .build ();

I have an infrared converter device, I found in the process of matching, ZWave Binding did not improve COMMAND_CLASS_SIMPLE_AV_CONTROL. I tried to improve this class myself and then found that adding the device after the generated xml file did not generate COMMAND_CLASS_SIMPLE_AV_CONTROL. Then I directly with a command to control the device, the following situation:
18: 27: 40.961 [INFO] [ommandClassTransactionPayloadBuilder] - commandClass Key 148
18: 27: 40.962 [INFO] [ommandClassTransactionPayloadBuilder] - At build null

I use the method is: new ZWaveCommandClassTransactionPayloadBuilder (getNode (). GetNodeId (), getCommandClass (), SIMPLE_AV_CONTROL_SET) .withPayload (outputData.toByteArray ()). WithPriority (TransactionPriority.Set) .build ();

Ask Mr. Chris to give corrections.
Thanks

cdjackson, 2017-10-31:
Sorry - I'm not really sure what you are asking. I guess it is not working, but there's no information that I can really comment on. If there's a log you have a question on, then please provide that. IF there's code you want me to comment on, then again, please provide and I'll be happy to have a look.

cdjackson, 2017-12-30:
Closing as no further information provided. If this is still relevant, please provide further information as requested a few months back.

closed 2017-12-30 by cdjackson