openhab/org.openhab.binding.zwave #730: ZHC5010 - offline communication error

SySfRaMe, 2017-11-03:
* OH2 Version: **2.1.0 release build**
* Binding Version: **zwave 2.1.0**
* Device name: **ZHC5010-WH-4L**
* Device version: **Configuration updated: 2017-03-12 14:35:03, firmware 1.09**
* Link to [database]: **http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/503**
* Device HW+SW: **Raspberry pi 3 with Raspbian**
* Zwave HW: **Zwave.me ZME usb**
* Distance between usb dongle and unit: **7m**
* XML file:
`<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>BINARY_SWITCH</genericDeviceClass>
    <specificDeviceClass>POWER_SWITCH_BINARY</specificDeviceClass>
  </deviceClass>
  <homeId>0xd0650336</homeId>
  <nodeId>13</nodeId>
  <version>4</version>
  <manufacturer>0x234</manufacturer>
  <deviceId>0x10a</deviceId>
  <deviceType>0x3</deviceType>
  <listening>true</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>true</beaming>
  <maxBaudRate>40000</maxBaudRate>
  <nodeInformationFrame>
    <commandClass>ZWAVE_PLUS_INFO</commandClass>
    <commandClass>VERSION</commandClass>
    <commandClass>MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>DEVICE_RESET_LOCALLY</commandClass>
    <commandClass>POWERLEVEL</commandClass>
    <commandClass>FIRMWARE_UPDATE_MD</commandClass>
    <commandClass>ASSOCIATION</commandClass>
    <commandClass>ASSOCIATION_GROUP_INFO</commandClass>
    <commandClass>MULTI_INSTANCE_ASSOCIATION</commandClass>
    <commandClass>CONFIGURATION</commandClass>
    <commandClass>MULTI_INSTANCE</commandClass>
    <commandClass>SWITCH_BINARY</commandClass>
    <commandClass>SWITCH_TOGGLE_BINARY</commandClass>
    <commandClass>SWITCH_MULTILEVEL</commandClass>
    <commandClass>CENTRAL_SCENE</commandClass>
    <commandClass>INDICATOR</commandClass>
  </nodeInformationFrame>
  <supportedCommandClasses>
    <entry>
      <commandClass>MULTI_INSTANCE_ASSOCIATION</commandClass>
      <multiAssociationCommandClass>
        <version>3</version>
        <instances>1</instances>
        <versionSupported>3</versionSupported>
        <maxGroups>21</maxGroups>
      </multiAssociationCommandClass>
    </entry>
    <entry>
      <commandClass>NO_OPERATION</commandClass>
      <noOperationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </noOperationCommandClass>
    </entry>
    <entry>
      <commandClass>FIRMWARE_UPDATE_MD</commandClass>
      <firmwareUpdateCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
      </firmwareUpdateCommandClass>
[... 1116 lines cut ...]
        <associations/>
      </associationGroup>
    </entry>
  </associationGroups>
  <nodeNeighbors/>
  <lastSent>2017-11-02 16:16:52.45 UTC</lastSent>
  <lastReceived>2017-11-02 16:16:47.155 UTC</lastReceived>
</node>`

*Log:
_20:27:29.901 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'zwave_device_2a4c54c2_node13_scene_number' received command 9.0
20:27:29.923 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_device_2a4c54c2_node13_scene_number changed from 2.0 to 9.0
20:27:36.219 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'zwave_device_2a4c54c2_node13_switch_binary2' received command ON
20:27:36.231 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_device_2a4c54c2_node13_switch_binary2 changed from OFF to ON
20:34:39.519 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:2a4c54c2:node13' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR): Node is not communicating with co         ntroller
20:34:56.217 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:2a4c54c2:node13' changed from OFFLINE (COMMUNICATION_ERROR): Node is not communicating with controller t         o ONLINE
20:35:14.891 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:2a4c54c2:node13' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR): Node is not communicating with co         ntroller
20:44:27.239 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:2a4c54c2:node13' changed from OFFLINE (COMMUNICATION_ERROR): Node is not communicating with controller to ONLINE
20:44:27.253 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_device_2a4c54c2_node13_switch_binary changed from ON to OFF
20:44:31.325 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_device_2a4c54c2_node13_switch_binary1 changed from ON to OFF
20:54:33.410 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:2a4c54c2:node13' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR): Node is not communicating with controller
20:54:33.426 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:2a4c54c2:node13' changed from OFFLINE (COMMUNICATION_ERROR): Node is not communicating with controller to ONLINE_

Best Regards
Lasse

cdjackson, 2017-11-03:
I'm not sure what the issue is you're reporting? I think it's probably best to discuss this on the forum as there will be a lot more people to help you and we can track the issue here once we establish there's a bug.

SySfRaMe, 2017-11-04:
Sorry mate, im new to this (-:
Where do i find the forum?
Should i leave this one open for now?

sihui62, 2017-11-05:
> Where do i find the forum?
https://community.openhab.org/

cdjackson, 2017-12-30:
Closing as I assume this is discussed elsewhere...

closed 2017-12-30 by cdjackson