openhab/org.openhab.binding.zwave #734: Exclude Device

Ynoy, 2017-11-07:
Hi Chris:
@cdjackson  @quadule @maxberger @kabili207 @asciiphil 
     I added more than twenty devices to my gateway and I found a problem removing the device from the Zwave network: via Paper UI's Exclude device, I stopped serial communication in a very short period of time and restarted Zwave Controller, which led to Exclude Device fails.If I move very fast, the device can be removed successfully. The device can successfully remove the network through HabMIN operation.Whatever one of these two approaches is adopted,  other devices will report the interruption of information. I can only control the device. For example, the triggering of the sensor device to report the status information and the Zwave Controller can not receive the information.
    May I ask how to solve this problem?

cdjackson, 2017-11-07:
Sorry, I'm not really sure what the problem is?

It sounds like you are removing the controller from the USB port and this is causing a problem which I can perfectly understand and it would be normal.

Your report is very confusion as you talk about exclusion, then you talk about device reporting status. Please can you clearly state what the bug is. Also, report what version of the binding you are using and what hardware. Do the other people listed above also have this problem (@quadule @maxberger @kabili207 @asciiphil)?

Ynoy, 2017-11-07:
I want to remove the device from the Zwave network. Instead of unplugging the USB Controller.
After removing the device from the Zwave network, the other devices are not communicating properly with the Zwave controller.All information about devices that are still in the Zwave network is not updated with the device's actual status.You must delete the device, re-add from Inbox, can communicate properly with Zwave Controller.(It looks like you need to reinitialize each Node)
I use org.openhab.binding.zwave-master (2.2.0).

cdjackson, 2017-11-07:
If you want to exclude a device then use the controller option in the thing menu in Habmin. If you remove the USB stick the. I sing will stop working. 

It is probably better to discuss your question on the forum rather than here.

closed 2017-11-15 by cdjackson