openhab/org.openhab.binding.zwave #737: Thermostat Mode value 15 (Boost) missing ?

chris922, 2017-11-11:
This issue list is to report bugs or enhancements for the openHAB Z-Wave binding. If you're not sure about something, please consider discussing it on the [forum](http://community.openhab.org) before raising an issue - more people follow the forum than monitor the issues list, so if you are having trouble with a device, or want to ask a question, you will have a much larger audience there.

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version: 2.1.0-1
* Binding Version: 2.2.0.201711040020

If the problem is associated with a device, please provide the following -:

* Device name: Spirit Z-Wave Plus
* Device version: firmware 0.15
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/710
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.
[node20.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1463703/node20.txt) _(as txt because xml is not allowed)_

Hi Chris + team,

I've got the mentioned device and this one sends a THERMOSTAT_MODE with the value 15 when we activate the boost mode and this is unknown. I also took a look at the code and there is no value in the ModeType enumeration for index 15.
https://github.com/openhab/org.openhab.binding.zwave/blob/01264939c156ed6e5d9e7e89ac511c6100eeedee/src/main/java/org/openhab/binding/zwave/internal/protocol/commandclass/ZWaveThermostatModeCommandClass.java#L282

Maybe the device is sending some wrong value or something needs to be changed to transform this value into the correct one, I don't know.

Could you please take a look at this? This prevents us from changing the mode for this device using openHAB.
In the manual chapter 6.2 are all possible modes listed:
http://www.cd-jackson.com/zwave_device_uploads/710/Spirit-Z-Wave-BAL-web-EN-view-04.pdf

Extract from the log, if you need more information please let me know.

	2017-11-11 12:21:21.981 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0A 00 04 00 14 04 40 03 0F 00 AD
	2017-11-11 12:21:22.015 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
	2017-11-11 12:21:22.018 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0A 00 04 00 14 04 40 03 0F 00 AD
	2017-11-11 12:21:22.020 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0A 00 04 00 14 04 40 03 0F 00 AD
	2017-11-11 12:21:22.022 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 14 04 40 03 0F 00
	2017-11-11 12:21:22.023 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 20: Application Command Request (ALIVE:DONE)
	2017-11-11 12:21:22.024 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 20: Starting initialisation from DONE
	2017-11-11 12:21:22.026 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@71246f already registered
	2017-11-11 12:21:22.027 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 20: Incoming command class THERMOSTAT_MODE
	2017-11-11 12:21:22.028 [DEBUG] [lass.ZWaveThermostatModeCommandClass] - NODE 20: Received Thermostat Mode Request
	2017-11-11 12:21:22.029 [DEBUG] [lass.ZWaveThermostatModeCommandClass] - NODE 20: Thermostat Mode report, value = 15
	2017-11-11 12:21:22.030 [DEBUG] [lass.ZWaveThermostatModeCommandClass] - NODE 20: Unknown Mode Type = 15, ignoring report.

Thanks for you support and best regards,
Christian

chris922, 2017-11-11:
I just added a pull request #738 that hopefully fixes this issue on the correct way. As I was not able to find any BOOST value in the enumeration and all other values from the enum matches the one described for this device I assume that the value 15 is the official one for BOOST.

cdjackson, 2017-11-11:
Closed by #738 
Thanks.

closed 2017-11-11 by cdjackson

chris922, 2017-11-11:
Thanks a lot for the fast reply! :+1: