openhab/org.openhab.binding.zwave #751: Binding for AlarmType.MOTION

jhmartin, 2017-11-20:
The ZSE40 is a ZWave Plus motion sensor (and other sensors). It is sending a notification that motion is detected, but this results in an exception that AlarmType.MOTION is not known:

* OH2 Version: 2.1

* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):
addons-2.1.0.

If the problem is associated with a device, please provide the following -:
Zooz ZSE40

* Device name: ZSE40 4-in-1 Sensor
* Device version: 
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/307

* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.
https://gist.github.com/jhmartin/4ed6b57493a3f19f81d431b483bbc8a7

```
2017-11-19 16:58:08.756 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.IllegalArgumentException: No enum constant org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass.AlarmType.MOTION
        at java.lang.Enum.valueOf(Unknown Source)[:1.8.0_152]
        at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass$AlarmType.valueOf(ZWaveAlarmCommandClass.java:1)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.converter.ZWaveAlarmConverter.handleNotificationReport(ZWaveAlarmConverter.java:144)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.converter.ZWaveAlarmConverter.handleEvent(ZWaveAlarmConverter.java:109)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.ZWaveIncomingEvent(ZWaveThingHandler.java:1099)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass.processNotificationReport(ZWaveAlarmCommandClass.java:194)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass.handleApplicationCommandRequest(ZWaveAlarmCommandClass.java:124)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:119)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[182:org.openhab.binding.zwave:2.1.0]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[182:org.openhab.binding.zwave:2.1.0]
```

cdjackson, 2017-11-20:
Please provide a debug file showing the problem.

jerryhurst, 2017-11-20:
I too am experiencing this and here's my DEBUG:

'''10:25:28.980 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 00 04 00 0B 03 20 01 00 DB 
10:25:29.020 [DEBUG] [ave.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
10:25:29.021 [DEBUG] [zwave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 04 00 0B 03 20 01 00 DB 
10:25:29.022 [DEBUG] [ave.internal.protocol.ZWaveController] - Process Message = 01 09 00 04 00 0B 03 20 01 00 DB 
10:25:29.022 [DEBUG] [ave.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 0B 03 20 01 00 
10:25:29.023 [DEBUG] [essage.ApplicationCommandMessageClass] - NODE 11: Application Command Request (ALIVE:DONE)
10:25:29.023 [DEBUG] [ialization.ZWaveNodeInitStageAdvancer] - NODE 11: Starting initialisation from DONE
10:25:29.023 [DEBUG] [ave.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@66aad01d already registered
10:25:29.023 [DEBUG] [essage.ApplicationCommandMessageClass] - NODE 11: Incoming command class BASIC
10:25:29.024 [DEBUG] [l.commandclass.ZWaveBasicCommandClass] - NODE 11: Received Basic Request
10:25:29.024 [DEBUG] [l.commandclass.ZWaveBasicCommandClass] - NODE 11: Basic Set sent to the controller will be processed as Basic Report
10:25:29.024 [DEBUG] [l.commandclass.ZWaveBasicCommandClass] - NODE 11: Basic report, value = 0x00
10:25:29.025 [DEBUG] [ave.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveCommandClassValueEvent
10:25:29.025 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 11: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
10:25:29.025 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 11: Got a value event from Z-Wave network, endpoint = 0, command class = BASIC, value = 0
10:25:29.026 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 11: Updating channel state zwave:device:da88d044:node11:sensor_binary to OFF [OnOffType]
10:25:29.026 [DEBUG] [essage.ApplicationCommandMessageClass] - NODE 10: Transaction not completed: node address inconsistent.  lastSent=10, incoming=255
10:25:29.028 [INFO ] [smarthome.event.ItemStateChangedEvent] - zooz_4n1_PIR_MOTION2 changed from ON to OFF
10:25:29.029 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 11 00 04 00 0B 0B 71 05 07 00 00 FF 07 00 01 08 00 68 
10:25:29.030 [DEBUG] [ave.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
10:25:29.030 [DEBUG] [zwave.internal.protocol.SerialMessage] - Assembled message buffer = 01 11 00 04 00 0B 0B 71 05 07 00 00 FF 07 00 01 08 00 68 
10:25:29.030 [DEBUG] [ave.internal.protocol.ZWaveController] - Process Message = 01 11 00 04 00 0B 0B 71 05 07 00 00 FF 07 00 01 08 00 68 
10:25:29.031 [DEBUG] [ave.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 0B 0B 71 05 07 00 00 FF 07 00 01 08 00 
10:25:29.031 [DEBUG] [essage.ApplicationCommandMessageClass] - NODE 11: Application Command Request (ALIVE:DONE)
10:25:29.031 [DEBUG] [ialization.ZWaveNodeInitStageAdvancer] - NODE 11: Starting initialisation from DONE
10:25:29.031 [DEBUG] [ave.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@66aad01d already registered
10:25:29.031 [DEBUG] [essage.ApplicationCommandMessageClass] - NODE 11: Incoming command class ALARM
10:25:29.032 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 11: Received ALARM command V4
10:25:29.032 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 11: Process NOTIFICATION_REPORT V4
10:25:29.032 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 11: NOTIFICATION report - 7 = 0, event=0, status=255
10:25:29.032 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 11: Alarm Type = BURGLAR (7)
10:25:29.032 [DEBUG] [ave.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
10:25:29.032 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 11: Got an event from Z-Wave network: ZWaveAlarmValueEvent
10:25:29.033 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 11: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
10:25:29.033 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
10:25:29.033 [ERROR] [ave.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.IllegalArgumentException: No enum constant org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass.AlarmType.MOTION
	at java.lang.Enum.valueOf(Enum.java:238) [?:?]
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveAlarmCommandClass$AlarmType.valueOf(ZWaveAlarmCommandClass.java:1) [223:org.openhab.binding.zwave:2.2.0.201711162213]
	at org.openhab.binding.zwave.internal.converter.ZWaveAlarmConverter.handleNotificationReport(ZWaveAlarmConverter.java:144) [223:org.openhab.binding.zwave:2.2.0.201711162213]
	at org.openhab.binding.zwave.internal.converter.ZWaveAlarmConverter.handleEvent(ZWaveAlarmConverter.java:109) [223:org.openhab.binding.zwave:2.2.0.201711162213]
[... cut ...]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248) [223:org.openhab.binding.zwave:2.2.0.201711162213]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214) [223:org.openhab.binding.zwave:2.2.0.201711162213]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208) [223:org.openhab.binding.zwave:2.2.0.201711162213]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324) [223:org.openhab.binding.zwave:2.2.0.201711162213]'''

cdjackson, 2017-11-20:
Thanks. I've updated the database and will update the binding in a few days - hopefully it will sove the issue.

cdjackson, 2017-12-30:
Closing as this should be fixed. If the problem persists with the latest binding please reopen.

closed 2017-12-30 by cdjackson

jerryhurst, 2018-01-01:
Confirmed this is fixed on my system, Thank you very much for your help.