openhab/org.openhab.binding.zwave #754: [DEV] Adding node_1_null

openhab-5iver, 2017-11-22:
https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653/1994?u=5iver

* OH2 Version: 2.2.0 snapshot build 1083
* Binding Version: 2.2.0.201711141803

`2017-11-14 13:56:01.566 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 42: Update ASSOCIATION group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@5052abb: Adding node_1_null`

I figured I'd write up some issues so that they don't get lost in the forum!

cdjackson, 2017-11-22:
I think that this is fixed in the latest version - can you try a more recent dev binding (or the version I posted yesterday)

openhab-5iver, 2017-11-22:
Same with 2.2.0.201711192127.

`zwave.log.50:2017-11-22 06:46:28.768 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 228: Update ASSOCIATION group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@1e6ddade: Adding node_1_null`

But the test jar you posted for someone else (2.2.0.201711211942) seems to have corrected this.

openhab-5iver, 2017-11-22:
In case edits don't send notifications... my grep scrolled the most recent off the screen and I missed it at first. Same result for 2.2.0.201711211942.

openhab-5iver, 2017-12-06:
Hey, Chris. Thought I'd bump this one. I still see node_1_null. I'm hoping the fix will also me to set associations for some locks that I had to reinclude.

cdjackson, 2017-12-09:
I think this is just a logging issue - does it actually cause any issues?

The code is here  -:
```
                            logger.debug("NODE {}: Update ASSOCIATION group_{}: Adding node_{}_{}", nodeId, groupId,
                                    groupMember.getNode(), groupMember.getEndpoint());
                            group.add(groupMember.toString());

```

So the problem is that groupMember.getEndpoint is returning null - this is fine in itself for nodes that don't support multiple endpoints. I'll update the logging, but it won't change anything. If there's another problem that you think exists then please advise.

openhab-5iver, 2017-12-09:
I have set the association multiple times for two locks that I had to reinclude, but it appears that the association is never set (never receive alarms). In looking at the nodes in Domoticz/OZW, these nodes did not show as associated to the controller, or even having a group. I waited a few hours but this never changed. My first assumption was that it was an association issue, maybe related to this issue, or maybe related to [755](https://github.com/openhab/org.openhab.binding.zwave/issues/755).

cdjackson, 2017-12-30:
@openhab-5iver I think this is a duplicate of #755. The above PR fixed the logging issue so I'll close this an we can deal with any further problems regarding setting associations in #755 .

closed 2017-12-30 by cdjackson