openhab/org.openhab.binding.zwave #757: [dev] Transaction retries

pizerg, 2017-11-27:
This might be intended behaviour but I wanted to make sure it's not a bug.

During some tests I started to see that some sensors (fibaro wall plug (versions with and without security)  and aeon smart switch 6) would ACK a SWITCH_BINARY_GET message but then it wouldn't send a SWITCH_BINARY_REPORT. Everything has been checked via zniffer + code logs.

As it's the sensor who is actively ignoring the command but sending the ACK the only option we have is to retry the GET message and here is when I'm not sure if the transaction is being handled properly. 

When the GET command is sent, a transaction is created with COMMAND_CLASS_SWITCH_BINARY as expected command class, so I assumed that if we do not receive the SWITCH_BINARY_REPORT before the timeout the transaction would be cancelled and there I could be able to handle the retries but before this happens the transaction is completed when a SendData message arrives with the proper callback ID and the ACK from the sensor.

The transaction is completed after the 3rd line in the following:

[image]

Shouldn't this transaction be completed when we have actually received the information we asked from the sensor?

PS. I know right now retries are not handled in the current version of the binding but I've modified it to use retries again as I consider that some cases are not fully covered by the controller's internal retries

Adrián

cdjackson, 2017-11-27:
Yes, the transaction for a GET should complete after the REPORT is received. This log doesn’t show the bindings transaction management though so we can’t confirm from this that it does (or doesn’t) do this.  The command generation is defined to do this at least -:

https://github.com/openhab/org.openhab.binding.zwave/blob/development/src/main/java/org/openhab/binding/zwave/internal/protocol/commandclass/ZWaveBinarySwitchCommandClass.java#L108 <https://github.com/openhab/org.openhab.binding.zwave/blob/development/src/main/java/org/openhab/binding/zwave/internal/protocol/commandclass/ZWaveBinarySwitchCommandClass.java#L108>

Can you provide the text log for this transaction as well (plus the 5 or 10 seconds after the times - up to 09:07:10 or so).

pizerg, 2017-11-28:
Here's the log, unfortunately I only got like an extra second or so, if needed I can try to reproduce the error and get some more logging, but you can see the transaction is completed and advanced to DONE status. You might see some extra logging I've added myself for debugging purposes.

[zw_log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1509582/zw_log.txt)

cdjackson, 2017-11-29:
I can't really tie up your log with the code - as a minimum there are debug messages missing that are in the code that are not in your log and without these I can't see why it's not working. It should work ok, and I'm reasonably confident that it does work on unmodified code - maybe there's some sort of corner case, but without the debug it's really hard to say much more.

Please can you ensure that the transaction code you're using is from the latest source and then provide another log.

pizerg, 2017-11-29:
Here's the log again, now with some more seconds after the issue, I can confirm that the transaction code is from the latest source I'd just forgotten to merge the logging. All logging messages should be there now. 

The affected node this time was NODE 2, a Fibaro Wall Plug.

[zw_log_corrected.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1514282/zw_log_corrected.txt)

pizerg, 2017-12-14:
Please let me know if there's still some information missing or if you need some extra data!

cdjackson, 2017-12-18:
Sorry. I've just not had the chance to look at this as I'm on holiday. I'll have a look once back in the UK in a week or so.

cdjackson, 2017-12-30:
The reason for this, is that depending on the method used to queue a request, the transaction handler may not wait for the response.  If the message is queued through the controller ```enqueue``` method, which in itself does not wait for a response, then the transaction is also set to not wait for a response to avoid having this call delay other transactions.

```
        // Since this method doesn't wait for a response, we tell the transaction handler not to wait for DATA
        if (transaction instanceof ZWaveCommandClassTransactionPayload) {
            ((ZWaveCommandClassTransactionPayload) transaction).setRequiresResponse(false);
        }

        transactionManager.queueTransactionForSend(transaction);
```

Is this actually causing problems, or was it just something that you saw that didn't make sense? I must admit I'd forgotten about this myself! ;)

pizerg, 2018-01-03:
Hi,

I'll try to explain the whole issue with full details to give a more accurate context:

This issue was reproduced when triggering multiple ON/OFF actions against different smart plug devices, each action generates two messages for each device (SET + GET), I noticed that sometimes one or two of the devices (tests run with 6 devices, a mix of fibaro and aeon smart plugs with and without security) were either not changing their state or changing it but not reporting back after they received the GET message. 

As I had some SECURITY sensors I thought that maybe there were too many messages generated and the controller/devices couldn't cope with them, so first thing I tried was to add a delay of 50ms between 
 messages, that seemed to improve the results as most of the time I was experiencing the expected behaviour, state change + state report but there were still some random errors. 

After further investigation using zniffer and the binding logging I noticed some issues with SECURITY sensors, some of them were resending the same nonce several times in a row (even after the controller ACK'd them) and that caused "controller to device" messages to be ignored when the binding sent two different messages using the same NONCE. To fix that I coded a simple workaround in ZWaveSecurityCommandClass that tracks the last NONCE ID received and ignores any further NONCE received with the same ID. That fixed the issue with some security messages being ignored by the sensors due to the same NONCE ID being used twice in a row.

After that I could still see random errors that happened when the device ACK'd a GET but refused to send the proper REPORT, as I assumed this was device related I thought the only way to go would be to implement software retries (as the controller retries did not help at all) and that was when I opened this issue.

So to finally answer your question I think that not waiting for the response is causing indirect issues because the transaction is marked as completed and ACK'd which causes the software retries code to never be triggered which causes the message that was ignored by the device not being resent.

PS. During the investigation of this issue I think I found another unrelated one but had no time to check / open a new issue, I think that queued messages for sleeping/battery devices do not have the desired priority (TransactionPriority.Immediate according to the documentation) so if you try to set/get/configure a sleeping device it might take several wake up notifications if there are other higher priority messages in the queue

cdjackson, 2018-01-03:
> To fix that I coded a simple workaround in ZWaveSecurityCommandClass that tracks the last NONCE ID received and ignores any further NONCE received with the same ID. 

So this just remembers the last NONCE ID and ignores any further request? Do you fancy providing a PR for this?

> So to finally answer your question I think that not waiting for the response is causing indirect issues because the transaction is marked as completed and ACK'd which causes the software retries code to never be triggered which causes the message that was ignored by the device not being resent.

Ok, so this issue is only something that's a problem in your configuration as you have retries enabled. In OH (at the moment) there are no retries so this isn't a problem.

Maybe I need to revisit the use of retries here. If so, then I agree that the transaction should probably wait for the response before the transaction is considered complete.

pizerg, 2018-01-04:
I've provided a PR for the NONCE issue

> Ok, so this issue is only something that's a problem in your configuration as you have retries enabled. In OH (at the moment) there are no retries so this isn't a problem.

The only visible problem will be the same issue that I experienced, in some cases there might be a missing GET REPORT if the sensor decides to ACK and ignore the command

cdjackson, 2018-01-04:
> I've provided a PR for the NONCE issue

Thanks.

> The only visible problem will be the same issue that I experienced, in some cases there might be a missing GET REPORT if the sensor decides to ACK and ignore the command

That's true, but my point is that if retries are disabled, then it will never retry anyway, and if the sensor decides to ignore the request after the ACK, then chances are it will ignore it again next time?

pizerg, 2018-01-04:
> That's true, but my point is that if retries are disabled, then it will never retry anyway, and if the sensor decides to ignore the request after the ACK, then chances are it will ignore it again next time?

I hoped that after retrying the message the device would respond as usual (I'm testing it later this morning, making the binding to wait for a response using a different sendData method).

From my experience the device would continue to work normally after it ignores a message as after reproducing the issue I could just request a state change and it would properly execute both SET + GET, so maybe this is all related to network congestion when I try to execute multiple commands very quickly. 

Will get back to you after I do further testing and can confirm this.

cdjackson, 2018-07-17:
Closing this since retries are now included in the dev binding. If there are still issues, please reopen, or open a new issue.

closed 2018-07-17 by cdjackson