openhab/org.openhab.binding.zwave #767: Binding throws exception on startup

kaikreuzer, 2017-12-14:
@cdjackson I am just doing final testing before the release.
Just by installing Z-Wave (not using it in any way), you will find this exception in your log:
```
15:36:48.767 [ERROR] [fig.xml.osgi.XmlDocumentBundleTracker] - Could not send adding finished event for the module 'org.openhab.binding.zwave'!
java.lang.IllegalArgumentException: UID must have at least 2 segments.
	at org.eclipse.smarthome.core.thing.UID.<init>(UID.java:70) [116:org.eclipse.smarthome.core.thing:0.10.0.201712141004]
	at org.eclipse.smarthome.core.thing.UID.<init>(UID.java:48) [116:org.eclipse.smarthome.core.thing:0.10.0.201712141004]
	at org.eclipse.smarthome.core.thing.type.ChannelTypeUID.<init>(ChannelTypeUID.java:40) [116:org.eclipse.smarthome.core.thing:0.10.0.201712141004]
	at org.eclipse.smarthome.core.thing.xml.internal.ChannelXmlResult.toChannelDefinition(ChannelXmlResult.java:121) [117:org.eclipse.smarthome.core.thing.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.toChannelDefinitions(ThingTypeXmlResult.java:96) [117:org.eclipse.smarthome.core.thing.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.getBuilder(ThingTypeXmlResult.java:148) [117:org.eclipse.smarthome.core.thing.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.toThingType(ThingTypeXmlResult.java:156) [117:org.eclipse.smarthome.core.thing.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlProvider.addingFinished(ThingTypeXmlProvider.java:151) [117:org.eclipse.smarthome.core.thing.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.addingFinished(XmlDocumentBundleTracker.java:218) [108:org.eclipse.smarthome.config.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.parseDocuments(XmlDocumentBundleTracker.java:358) [108:org.eclipse.smarthome.config.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.processBundle(XmlDocumentBundleTracker.java:336) [108:org.eclipse.smarthome.config.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.access$3(XmlDocumentBundleTracker.java:331) [108:org.eclipse.smarthome.config.xml:0.10.0.201712141004]
	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker$2.run(XmlDocumentBundleTracker.java:307) [108:org.eclipse.smarthome.config.xml:0.10.0.201712141004]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) [?:?]
	at java.lang.Thread.run(Thread.java:745) [?:?]
```
I have tracked this down to [this line](https://github.com/openhab/org.openhab.binding.zwave/blob/master/ESH-INF/thing/hank_acc01_0_0.xml#L14), where no channel type id is provided. I am not sure if there are other such places as well, but it seems that your latest database export was corrupt.

Please fix asap for the release! Thanks.

martinvw, 2017-12-14:
@cdjackson is this maybe the cause of the problem, its not yet approved, so should it have been skipped in exporting:

> Approved on -0001-11-30 00:00:00, and last exported on 2017-12-11 21:02:51.

cdjackson, 2017-12-14:
Approved/exported should be the same - it’s the same button in the database. I’ve removed this file anyway.

I’m updating the database tonight - it’s a little slow as I’m in the Central American rainforest, but I’ll try and do it today/tonight.

martinvw, 2017-12-14:
> it’s a little slow as I’m in the Central American rainforest, but I’ll try and do it today/tonight.

That is a pity 😉 Have a nice stay there and thanks for your quick reply / action!

kaikreuzer, 2017-12-14:
Yep, thanks @cdjackson for saving the release from your remote holiday destination 👍

cdjackson, 2017-12-14:
No probs - it’s amazing there’s even internet here ;)

I see reports on the forum that this is confirmed as fixed…

Chris

kaikreuzer, 2017-12-14:
Excellent, thanks!

closed 2017-12-14 by kaikreuzer