openhab/org.openhab.binding.zwave #778: [DEV] Save not enabled for updating zWave Device Configs

travellingkiwi, 2017-12-27:
This issue list is to report bugs or enhancements for the openHAB Z-Wave binding. If you're not sure about something, please consider discussing it on the [forum](http://community.openhab.org) before raising an issue - more people follow the forum than monitor the issues list, so if you are having trouble with a device, or want to ask a question, you will have a much larger audience there.

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version: 2.3.0-snapshot (Build #1151)
* Binding Version : 2.3.0.201712260145

If the problem is associated with a device, please provide the following -:

* Device name: Aeotec Multi Sensor 6

Not limited to the Aeotec multi sensor 6, but I just happen to have 2 of them I can't alter the config for.

When opening the config edit, the save (Blue Tick Button) at the top is disabled. Sometimes I find an existing setting that OH says is invalid and updating that to a 'supported' value enables the save. However on several devices there is no setting highlighted in red to indicate what the save is disabled.

Can we have something that makes it obvious what the issue is stopping the save? Especially since the settings currently in use on those devices are the DEFAULT ones that something (Presumably the zWave binding) initialised.

Note that although raising this against the current DEV (i.e. 2.3.0 snapshot) the same issue affects 2.2.0-1 release.

travellingkiwi, 2017-12-27:
Note that on one of them, the 'Temperature Threshold' might be responsible. Initially it's blank. And save is Disabled

[image]

Inserting ANYTHING in there appears to 'enable' the save...

[image]

However unlike other devices where I've had this issue, the default (blank) value wasn't highlighted as an issue (i.e. in Red like some devices)

cdjackson, 2017-12-27:
I guess there are two issues here, and it’s not directly related to the binding so I’ll close this.

First issue - you want something to make it obvious what is stopping the save - then this is a UI issue and I’d suggest to raise it in the UI you are using (PaperUI?)

The second is if there are issues with the database, then please get an account at the database site (www.cd-jackson.com) and help fix them.

Cheers
Chris

closed 2017-12-27 by cdjackson