openhab/org.openhab.binding.zwave #786: [DEV] Incorrect nodes reported by controller

openhab-5iver, 2017-12-30:
* OH2 Version: OH 2.3 snapshot 1149
* Binding Version: 2.2.0.201712301543 
* controller: HUSBZB-1 (occurs on two of them)

I'm getting some log entries for nodes that are not on the network, and some are from node numbers that are not possible. I've had 31 of them today (some are from the same node). There appear to be more of these during heavier traffic (binding restart), but they occur at other times too. I don't see this causing issue, because it looks like it is being well handled by the binding, but maybe there's something more to it and nobody has noticed. It could also be a hardware or firmware bug in this controller. It may just be time to rebuild my zwave network.

```
zwave.log.1:2017-12-30 14:22:40.281 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 182: Not initialized (ie node unknown), ignoring message.
zwave.log.1:2017-12-30 10:25:42.978 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 248: Not initialized (ie node unknown), ignoring message.
zwave.log.1:2017-12-30 11:08:52.038 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 250: Not initialized (ie node unknown), ignoring message.
```

Here is a full log from a previous occurrence. It seems like the event before and after are related, but node 250 is obviously not right. There always seems to be similar communications at the times this occurs, but the other device is always different (not node 186). If I get an urge, I will restore the zwave network to an Aeon zstick and see if the same happens on different hardware. I have an open warranty issue with the store that sold me the HUSBZB-1 for an issue where nodes just disappear, but after removing>resetting>reincluding the device the old node shows back up again. I guess it's possible this could be in some way related.

```
2017-12-24 01:38:28.952 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 08 00 04 10 BA 02 98 40 83 
2017-12-24 01:38:28.956 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Response SENT 6
2017-12-24 01:38:28.957 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=186, callback=16, payload=10 BA 02 98 40 
2017-12-24 01:38:28.957 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=186, callback=16, payload=10 BA 02 98 40 
2017-12-24 01:38:28.957 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=186, callback=16, payload=10 BA 02 98 40 
2017-12-24 01:38:28.958 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - lastTransaction TID 1097: [WAIT_DATA] requiresResponse=true callback: 0
2017-12-24 01:38:28.958 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Application Command Request (ALIVE:DONE)
2017-12-24 01:38:28.958 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 186: resetResendCount initComplete=true isDead=false
2017-12-24 01:38:28.958 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 186: Incoming command class COMMAND_CLASS_SECURITY, endpoint 0
2017-12-24 01:38:28.958 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 186: SECURITY check internal
2017-12-24 01:38:28.958 [DEBUG] [e.internal.protocol.commandclass.ZWaveCommandClass] - NODE 186: Received COMMAND_CLASS_SECURITY V1 SECURITY_NONCE_GET
2017-12-24 01:38:28.958 [DEBUG] [.protocol.commandclass.impl.CommandClassSecurityV1] - Creating command message SECURITY_NONCE_REPORT version 1
2017-12-24 01:38:28.958 [DEBUG] [saction.ZWaveCommandClassTransactionPayloadBuilder] - At build null
2017-12-24 01:38:28.958 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Added to secure queue - size 13
2017-12-24 01:38:28.958 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Commands processed 1.
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@3d1882f3.
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Checking transaction 1097  ApplicationUpdate.
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Checking transaction : state >> WAIT_DATA
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Checking transaction : node  >> 113
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Checking transaction : class >> 152 == null.
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Checking transaction : commd >> 64 == null.
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Command NOT verified org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@3d1882f3.
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 1
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 1
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction lastTransaction outstanding...
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - STOP transaction timer
2017-12-24 01:38:28.959 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Start transaction timer to Sun Dec 24 01:38:40 EST 2017 - 11140ms
2017-12-24 01:38:29.038 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 08 00 04 10 FA 02 98 40 C3 
2017-12-24 01:38:29.042 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Response SENT 6
2017-12-24 01:38:29.043 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=250, callback=16, payload=10 FA 02 98 40 
2017-12-24 01:38:29.043 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=250, callback=16, payload=10 FA 02 98 40 
2017-12-24 01:38:29.043 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=250, callback=16, payload=10 FA 02 98 40 
2017-12-24 01:38:29.043 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - lastTransaction TID 1097: [WAIT_DATA] requiresResponse=true callback: 0
2017-12-24 01:38:29.044 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 250: Not initialized (ie node unknown), ignoring message.
2017-12-24 01:38:29.044 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2017-12-24 01:38:29.044 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start
2017-12-24 01:38:29.044 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction lastTransaction outstanding...
2017-12-24 01:38:29.044 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - STOP transaction timer
2017-12-24 01:38:29.044 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Start transaction timer to Sun Dec 24 01:38:40 EST 2017 - 11055ms
2017-12-24 01:38:29.502 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 08 00 04 10 BA 02 98 40 83 
2017-12-24 01:38:29.506 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Response SENT 6
2017-12-24 01:38:29.507 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=186, callback=16, payload=10 BA 02 98 40 
[... cut ...]
2017-12-24 01:38:29.510 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 186: Command NOT verified org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@30555a3b.
2017-12-24 01:38:29.510 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 1
2017-12-24 01:38:29.510 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 1
2017-12-24 01:38:29.510 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
```

cdjackson, 2017-12-30:
Well, this is indeed pretty interesting -:

```
2017-12-24 01:38:29.038 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 08 00 04 10 FA 02 98 40 C3 
2017-12-24 01:38:29.043 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], dest=250, callback=16, payload=10 FA 02 98 40 
2017-12-24 01:38:29.044 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 250: Not initialized (ie node unknown), ignoring message.
```

This shows a valid frame (CRC check etc) with what looks like a valid command (it's a security NONCE_GET command - the 98 40).

It's probably meant to have come from node BA (so 1 bit flipped) but the checksum on the end of the frame is also consistent, so it's not an error coming from the serial port receive - unless you think that it's possible for the checksum to have also flipped consistently with the other bit! I think this is extremely improbable ;)

So, I don't see an issue with the binding here, but the controller seems to be sending rubbish... I'd suggest it's time to reset it!

cdjackson, 2017-12-31:
@openhab-5iver I'm tempted to close this as I'm pretty sure it's an issue with the stick itself. You ok with that or do you have further thoughts?....

openhab-5iver, 2017-12-31:
Yes, this can be closed. I'm still doing some more investigation and I'll reopen if I see anything that points to something other than hardware. Thank you for confirming. One question though... do you think this could possibly be caused, or aggravated, by the baud or flow control being used? I still haven't gotten any good information out of the manufacturer on recommended settings. Maybe this would go away or improve if the baud was lowered? With Zwave and Zigbee both running on the stick, it's got to be pretty busy. Makes me wonder if something similar is happening on the Zigbee side, but it is communicating at half the speed (baud is 57600).

Anyhow, I ran this on my backups...

egrep -r "Not initialized|Receive Message" >/opt/openhab2/notinitialized.log

... and I'm reading through the log to see if it just looks random, or if there is any consistency. This does seem to happen slightly more often during communications with my locks, and it's odd that the incorrect node is sometimes sent multiple times.

```
zwave.log.23:2017-12-31 05:58:56.637 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 DD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 45 
zwave.log.23:2017-12-31 05:58:56.744 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:56.751 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
zwave.log.23:2017-12-31 05:58:56.756 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:56.762 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
zwave.log.23:2017-12-31 05:58:56.767 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 DD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 45 
zwave.log.23:2017-12-31 05:58:56.807 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 DD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 45 
zwave.log.23:2017-12-31 05:58:56.858 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:56.864 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
zwave.log.23:2017-12-31 05:58:56.883 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:56.889 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
zwave.log.23:2017-12-31 05:58:56.894 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:56.898 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
zwave.log.23:2017-12-31 05:58:56.942 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 DD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 45 
zwave.log.23:2017-12-31 05:58:57.056 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:57.063 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
zwave.log.23:2017-12-31 05:58:57.176 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:57.182 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
zwave.log.23:2017-12-31 05:58:57.372 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 DD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 45 
zwave.log.23:2017-12-31 05:58:57.677 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 15 00 04 10 CD 0F 32 02 21 44 00 00 00 07 00 3C 00 00 00 07 00 55 
zwave.log.23:2017-12-31 05:58:57.685 [WARN ] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 205: Not initialized (ie node unknown), ignoring message.
```

cdjackson, 2018-01-01:
> do you think this could possibly be caused, or aggravated, by the baud or flow control being used?
I don't think so - I don't believe that this is any sort of corruption in the serial link - the software in the stock will have put a checksum on this frame, and the binding confirms the checksum is correct, so it's not a corruption there.

Also, it's very systematic - ie the node is "always" the same. In the grep above it's always changing from DD to CD - you'd expect most errors to be more random...

Given that at least some of these node ids are outside of the allowable range, it's a little surprising to see them come through...

I suspect the manufacturer won't have a lot to say as the zwave element is probably a standard Sigma chip/firmware, but worth trying. Maybe there's something related to the backup/restore that I think you said you did?

And of course, happy new year...

I'll close, but if you do hear back from anyone about this, feel fee to update.

closed 2018-01-01 by cdjackson