openhab/org.openhab.binding.zwave #789: Fibaro Roller Shutter 2 is not recognized correctly anymore

arenk, 2018-01-02:
The Fibaro Roller Shutter 2 does not work anymore following the update 4 days ago (Database update #780). I cannot change the parameters in PaperUI (no parameters are shown) anymore and the node is grayed out in HABmin.

The only difference I could see comparing the two xml-versions: Using FGR222 instead of FGRM222 as before (thingID and device property -> modelId)

Could this be an issue? I'm not too familiar with Z-Wave definitions.

cdjackson, 2018-01-02:
The change of name should not make any difference. The change in thing type should sort itself out unless you are using text definitions. If not, I would suggest to delete the thing and add it back to it definitely picks up the correct definition.

arenk, 2018-01-02:
Thanks for your quick reply! Deleting and re-adding the thing solved the issue!

closed 2018-01-02 by arenk