openhab/org.openhab.binding.zwave #793: Add configuration parameter option for Qubino Dimmer

mplutka, 2018-01-05:
* OH2 Version: 2.2.0

If the problem is associated with a device, please provide the following -:

* Device name: ZMNHDD Flush Dimmer Plus

Config item 20 has a third option, so the available options should look like this:
`        <options>
          <option value="0">0 - single push button (connected to I1)</option>
          <option value="1">1 - 3-way switch (connected to I1 and I2)</option>
          <option value="2">2 - Additional switch connected to I2</option>
        </options>
`

cdjackson, 2018-01-05:
Hi,
That doesn't appear to be what the manual states - at least not the version that is currently in the database (http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/210). Maybe this is for a later version of the firmware?

Please can you update this on the database site. If there is a newer version of the firmware that has this option, then it should be fine to just add this extra option, but please also upload the latest manual so that it's consistent.

Thanks
Chris

mplutka, 2018-01-05:
Hi, 
I don't have edit rights for the database but there is indeed a new version of the firmware/manual available (1.9): http://qubino.com/download/990/

Thanks
Markus

openhab-5iver, 2018-01-05:
@mplutka, I will add this configuration parameter for you, but could you please check to see if this is the only change in the newer firmware?

mplutka, 2018-01-05:
I can't see any other changes to configuration parameters than the one I
mentioned.

openhab-5iver, 2018-01-05:
OK... thank you for checking.

cdjackson, 2018-01-05:
> I don't have edit rights for the database

I've updated this.

@openhab-5iver thanks for supporting.

cdjackson, 2018-01-05:
It looks like the update is done so I'll close this. I'll probably do an update to the binding tomorrow.

closed 2018-01-05 by cdjackson