openhab/org.openhab.binding.zwave #795: [DEV] Enable SUC in default controller config

openhab-5iver, 2018-01-06:
https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653/2305?u=5iver

https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653/2307?u=5iver

As reported in forum, the default controller settings have SUC disabled. Most people would be setting up systems with a single controller, so I believe SUC should be enabled by default.

cdjackson, 2018-03-11:
Should now be implemented

closed 2018-03-11 by cdjackson

openhab-5iver, 2018-03-11:
Habmin displays this as a 1. Should I open a feature request to change that to Yes/No?

[image]

cdjackson, 2018-03-11:
No - this is a node number…  It allows you to set the node of the SIS and it defaults to the local controller if it’s not set.

openhab-5iver, 2018-03-11:
Ahhhhh. I read too fast.