openhab/org.openhab.binding.zwave #797: java.lang.NumberFormatException configuring Association Group

dude0001, 2018-01-08:
I am on the latest release of OH and the ZWave binding. I redeployed the latest version yesterday to fix a NULL pointer error when updating a Thing similar to what is describe in this [issue](https://github.com/openhab/org.openhab.binding.zwave/issues/755)

I can trace through the code and see the exception

> java.lang.NumberFormatException: For input string: "org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@3dd05f63"

is happening [here](https://github.com/openhab/org.openhab.binding.zwave/blob/5f789173512d24ec2a9168557d8701d3dbe8bdad/src/main/java/org/openhab/binding/zwave/handler/ZWaveThingHandler.java#L608). The code is seeing this is a "group" configuration and expecting the second argument in the configurationParameter to be a groupId and an integer. The code is trying to parse this long string as an Integer which of course fails.  However, I am a OH and ZWave newb and I am not sure what the correct solution should be. I'm also not sure if this is a problem with the code linked in the handler, or if the UI (I am using Paper UI) is passing a bad parameter here.

I am a software developer (C#), so I should be able to help troubleshoot quickly. I'd love to contribute myself, but I'm not sure how to setup a development environment to play with the code yet. Please let me know if you need more information.

Here is the HTTP PUT from Paper UI

> curl "http://megaman:8181/rest/things/zwave:device:e7d0219a:node7/config" -X PUT -H "Origin: http://megaman:8181" -H "Accept-Encoding: gzip, deflate" -H "Accept-Language: en-US,en;q=0.9" -H "User-Agent: Mozilla/5.0 (Linux; Android 6.0; Nexus 5 Build/MRA58N) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.108 Mobile Safari/537.36" -H "Content-Type: application/json" -H "Accept: application/json, text/plain, */*" -H "Referer: http://megaman:8181/paperui/index.html" -H "Connection: keep-alive" --data-binary "^{^\^"config_11_1^\^":1,^\^"config_10_1^\^":3,^\^"group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup^@3dd05f63^\^":^[^],^\^"group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup^@55eec832^\^":^[^],^\^"group_1^\^":^[^\^"node_1_0^\^"^],^\^"group_3^\^":null,^\^"group_2^\^":^[^],^\^"switchall_mode^\^":255,^\^"action_reinit^\^":false,^\^"group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup^@2d79c02f^\^":^[^\^"node_1_0^\^"^],^\^"config_12_1^\^":3,^\^"group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup^@2fbe54b7^\^":^[^\^"node_1_0^\^"^],^\^"action_failed^\^":false,^\^"powerlevel_level^\^":0,^\^"action_remove^\^":false,^\^"binding_pollperiod^\^":1800,^\^"action_heal^\^":false,^\^"group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup^@66bfec4^\^":^[^],^\^"config_7_1^\^":1,^\^"config_8_1^\^":3,^\^"config_9_1^\^":1,^\^"group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup^@623f221^\^":^[^],^\^"powerlevel_timeout^\^":0,^\^"config_3_1^\^":0,^\^"config_4_1^\^":0,^\^"config_5_1^\^":0^}" --compressed &

Here is the ZWave trace log

> 08-Jan-2018 10:44:50.367 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 7: Configuration update received
08-Jan-2018 10:44:50.367 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 7: Configuration update config_11_1 to 1
08-Jan-2018 10:44:50.367 [DEBUG] [otocol.commandclass.ZWaveConfigurationCommandClass] - NODE 7: Creating new message for application command CONFIGURATIONCMD_SET
08-Jan-2018 10:44:50.367 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - NODE 7: Creating empty message of class = SendData (0x13), type = Request (0x00)
08-Jan-2018 10:44:50.367 [TRACE] [ab.binding.zwave.internal.protocol.ZWaveController] - Callback ID = 77
08-Jan-2018 10:44:50.367 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Message queued. Queue length = 1. Queue={}
08-Jan-2018 10:44:50.367 [DEBUG] [otocol.commandclass.ZWaveConfigurationCommandClass] - NODE 7: Creating new message for application command CONFIGURATIONCMD_GET
08-Jan-2018 10:44:50.367 [DEBUG] [.internal.protocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 0
08-Jan-2018 10:44:50.367 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - NODE 7: Creating empty message of class = SendData (0x13), type = Request (0x00)
08-Jan-2018 10:44:50.368 [TRACE] [ab.binding.zwave.internal.protocol.ZWaveController] - Callback ID = 78
08-Jan-2018 10:44:50.368 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - Calculated checksum = 0xF5
08-Jan-2018 10:44:50.368 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Message queued. Queue length = 1. Queue={}
08-Jan-2018 10:44:50.368 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 7: Configuration update config_10_1 to 3
08-Jan-2018 10:44:50.368 [DEBUG] [nhab.binding.zwave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0C 00 13 07 05 70 04 0B 01 01 25 4D F5 
08-Jan-2018 10:44:50.368 [DEBUG] [otocol.commandclass.ZWaveConfigurationCommandClass] - NODE 7: Creating new message for application command CONFIGURATIONCMD_SET
08-Jan-2018 10:44:50.368 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - NODE 7: Creating empty message of class = SendData (0x13), type = Request (0x00)
08-Jan-2018 10:44:50.368 [DEBUG] [g.openhab.binding.zwave.handler.ZWaveSerialHandler] - NODE 7: Sending REQUEST Message = 01 0C 00 13 07 05 70 04 0B 01 01 25 4D F5 
08-Jan-2018 10:44:50.368 [TRACE] [ab.binding.zwave.internal.protocol.ZWaveController] - Callback ID = 79
08-Jan-2018 10:44:50.368 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Message queued. Queue length = 2. Queue={}
08-Jan-2018 10:44:50.368 [DEBUG] [otocol.commandclass.ZWaveConfigurationCommandClass] - NODE 7: Creating new message for application command CONFIGURATIONCMD_GET
08-Jan-2018 10:44:50.368 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - NODE 7: Creating empty message of class = SendData (0x13), type = Request (0x00)
08-Jan-2018 10:44:50.368 [TRACE] [ab.binding.zwave.internal.protocol.ZWaveController] - Callback ID = 80
08-Jan-2018 10:44:50.368 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Message queued. Queue length = 3. Queue={}
08-Jan-2018 10:44:50.368 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 7: Configuration update group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@3dd05f63 to []
08-Jan-2018 10:44:50.368 [TRACE] [g.openhab.binding.zwave.handler.ZWaveSerialHandler] - Message SENT
08-Jan-2018 10:44:50.369 [TRACE] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Received ACK
08-Jan-2018 10:44:50.375 [TRACE] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Received SOF
08-Jan-2018 10:44:50.375 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8 
08-Jan-2018 10:44:50.375 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - NODE 255: Creating new SerialMessage from buffer = 01 04 01 13 01 E8 
08-Jan-2018 10:44:50.375 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - Calculated checksum = 0xE8
08-Jan-2018 10:44:50.375 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - NODE 255: Checksum matched
08-Jan-2018 10:44:50.375 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - NODE 255: Message payload = 01 
08-Jan-2018 10:44:50.375 [TRACE] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Message is valid, sending ACK
08-Jan-2018 10:44:50.375 [TRACE] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Response SENT
08-Jan-2018 10:44:50.375 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
08-Jan-2018 10:44:50.376 [TRACE] [nhab.binding.zwave.internal.protocol.SerialMessage] - Calculated checksum = 0xE8
08-Jan-2018 10:44:50.376 [DEBUG] [nhab.binding.zwave.internal.protocol.SerialMessage] - Assembled message buffer = 01 04 01 13 01 E8 
08-Jan-2018 10:44:50.376 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Process Message = 01 04 01 13 01 E8 
08-Jan-2018 10:44:50.376 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Response[0x01], priority=High, dest=255, callback=0, payload=01 
08-Jan-2018 10:44:50.376 [TRACE] [ab.binding.zwave.internal.protocol.ZWaveController] - Incoming Message type = RESPONSE
[... cut ...]
INFO   | jvm 1    | 2018/01/08 09:51:59 | 	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run(ExecuteProduceConsume.java:136) [90:org.eclipse.jetty.util:9.3.22.v20171030]
INFO   | jvm 1    | 2018/01/08 09:51:59 | 	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:671) [90:org.eclipse.jetty.util:9.3.22.v20171030]
INFO   | jvm 1    | 2018/01/08 09:51:59 | 	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:589) [90:org.eclipse.jetty.util:9.3.22.v20171030]
INFO   | jvm 1    | 2018/01/08 09:51:59 | 	at java.lang.Thread.run(Thread.java:748) [?:?]

cdjackson, 2018-01-08:
Please consider trying the development version that is fixed as per the link you posted. Fixes are currently going into this branch.

https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653 <https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653>

dude0001, 2018-01-08:
@cdjackson 
The comment I made on [Issue #796](https://github.com/openhab/org.openhab.binding.zwave/issues/796) was regarding a NullPointerException I believe was already found in [Issue #755](https://github.com/openhab/org.openhab.binding.zwave/issues/755) and addressed with [Pull Request #758](https://github.com/openhab/org.openhab.binding.zwave/pull/758).

The issue above is a NumberFormatException and I believe a different issue. I haven't see it discussed elsewhere. Which link are you referring to exactly? Are you saying you believe this NumberFormatException may already be fixed in the development branch as well?

cdjackson, 2018-01-08:
The development branch is a very different source base so I don’t event know if it exists in the development version. The current master will be switched to the dev version in the near future and all fixes are going into that version.

dude0001, 2018-01-08:
Just browsing the [develop branch](https://github.com/openhab/org.openhab.binding.zwave/blob/45f23a789d061361a42a1fa9e64f884dd8a2261f/src/main/java/org/openhab/binding/zwave/handler/ZWaveThingHandler.java#L666), the code that is causing the NumberFormatException with a "group" configurationParameter looks almost identical. I don't see anything added that would not cause the NumberFormatException with the congigurationParameter

> group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@623f221

The code seems to split the congigurationParameter on underscore characters, switch on value on index 0 which is "group" here, and then the case for "group" takes the value at index 1 "org.openhab....." here and tries to parse it as an integer.

I'm willing to give it a try though. I've very new to the community. Just to clarify;

I would need to deploy just the ZWave Binding from the development branch build, and I can use the current OH 2.2.0 release I got from [here](https://www.openhab.org/downloads.html) this morning?

Are there any instructions for this?

cdjackson, 2018-01-08:
Yes - you can use the standard OH2.2 runtime.  Take a look at the top half dozen messages in the thread I linked - it talks about how to deploy. I’ll take a look at this in the dev branch although slightly strange it’s not been reported if there’s a problem there.

cdjackson, 2018-01-08:
I don't see anything wrong with the code you linked -:

```
                    if (cfg.length < 2) {
                        logger.debug("NODE{}: Association invalid {}", nodeId, configurationParameter.getKey());
                        continue;
                    }

                    Integer groupIndex = Integer.valueOf(cfg[1]);
```

It does make an assumption that cfg[1] is an Integer, but there should be no way that this can not be an integer unless the UI is somehow sending an ill formed command.

dude0001, 2018-01-08:
That is what it is sending if you look at the logs above. I also admitted above I'm a complete newb to OH and it very well could be an issue with my setup or that Paper UI is sending a bad or unexpected parameter here.

I'm using GE ZWave Dimmers that have ZWave Plus. I've read some other things I don't fully understand that there have been some issues with Group Associations with GE's implementation of ZWave Plus. No idea if they are related either. My gut says it is something related to this though.

After reading through your thread some more, on a whim I deleted the Thing and added them back from the Inbox, but still have the same issue. I'm going to work on deploying the development branch build. 

Any other information you think could help here? It looks like in your thread people are using a different UI than Paper UI. I could give that one a try if you point me to it as well.

cdjackson, 2018-01-08:
What devices are you using this with?  It won't be anything to do with GEs implementation though as the data doesn't come from the device.

It's difficult to view your data - please use code fences in future to improve the formatting.

dude0001, 2018-01-08:
I updated my OP so the logs are in code fences instead of quotes, sorry about that.

Here is a screen shot of one of the GE ZWave dimmers I'm having the issue with. You can see the 500 error as I had just tried to save the Thing to add a location and turn the night light option off (parameter 3). The location saved but the night light configuration does not change.. I have some dimmers and on/off switches that all have the problem.

[image]

cdjackson, 2018-01-08:
So the problem is this line -:

```
08-Jan-2018 10:44:50.517 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 7: Processing group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@623f221
```

Not sure why PaperUI is sending this. Possibly something was saved in an invalid way earlier, and now its in the config.

I can of course protect against this, but I don't like adding loads of checks that aren't needed as it just slows things down and makes the code less readable. At the end of the day, this is an invalid configuration.

Using HABmin will also solve this problem by the way.

dude0001, 2018-01-08:
I agree that seems to be the line in the debug log I'm tracing the code with and where the exception comes from. I don't know more about this all right now to know how to troubleshoot more, but I agree it could be from the UI side.

With that said, I think you agree the develop branch code doesn't do anything to address this. So I'm going to wait to deploy it for now. I'm trying to learn and throwing an unstable binding in to the mix might not be beneficial for me unless you want the feedback.

Is Paper UI not preferred? I haven't tried HABmin yet but seen it referenced. I could give that a try.

When you say something was saved in an invalid way earlier: how would you suggest I "resave" it to make sure it is saved in the correct format?  I did try deleting the Thing and readding it through Inbox. Are you suggesting to disassociate the devices and completely readd them etc?

cdjackson, 2018-01-08:
No - I didn’t say that the dev version doesn’t address this. I don’t thin this will occur in the dev version. The exception might not be caught, but it’s also possible that the configuration issue would not get into this state in the dev version in the first place.

PaperUI sends ALL parameters when the configuration is updated - HABmin will only send the parameters that have changed, so I’m pretty sure that will stop this.

If you change to the dev version, you will need to delete all things and add them back (that doesn’t mean exclude/include). That will remove the old configuration.

dude0001, 2018-01-08:
I was reading too much in to your response.  uninstalled the ZWave binding, and manually deployed the development branch snapshot build. You are correct, the error is gone now. I can set the night light and other parameters now.

I'll mark this as closed. I agree with you, I wouldn't make an additional code change.  Thank you for the help. I look forward to contributing in the future.

BTW, I also looked at HABmin as well. I didn't realize you were the owner of that project too.  It definitely looks nice and fun to play with. I'll probably give it a try after I finish some other projects.

closed 2018-01-08 by dude0001