openhab/org.openhab.binding.zwave #802: [DEV] Exception when manually starting a heal on specific devices (DZMX1 and DZS15)

openhab-5iver, 2018-01-13:
* OH2 Version: 2.3.0 snaphot 1177
* Binding Version: 2.2.0.201801021718 
* Device name: Leviton [DZS15](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/195) and [DZMX1](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/196)

Following up from [this request](https://github.com/openhab/org.openhab.binding.zwave/issues/755#issuecomment-354539150), when trying to heal these devices, Habmin displays _Error sending action_ and the following exception comes up in the log...

```
2018-01-13 05:33:21.801 [ERROR] [marthome.io.rest.core.internal.thing.ThingResource] - Exception during HTTP PUT request for update config at 'things/zwave:device:07cb40a2:node6/config'
java.lang.NullPointerException: null
        at org.eclipse.smarthome.config.xml.AbstractXmlBasedProvider.get(AbstractXmlBasedProvider.java:151) [110:org.eclipse.smarthome.config.xml:0.10.0.201712141004]
        at org.eclipse.smarthome.config.xml.AbstractXmlConfigDescriptionProvider.getConfigDescription(AbstractXmlConfigDescriptionProvider.java:50) [110:org.eclipse.smarthome.config.xml:0.10.0.201712141004]
        at org.eclipse.smarthome.config.core.ConfigDescriptionRegistry.getConfigDescription(ConfigDescriptionRegistry.java:152) [105:org.eclipse.smarthome.config.core:0.10.0.201712141004]
        at org.eclipse.smarthome.config.core.ConfigDescriptionRegistry.getConfigDescription(ConfigDescriptionRegistry.java:188) [105:org.eclipse.smarthome.config.core:0.10.0.201712141004]
        at org.eclipse.smarthome.io.rest.core.internal.thing.ThingResource.normalizeConfiguration(ThingResource.java:735) [128:org.eclipse.smarthome.io.rest.core:0.10.0.201712141004]
        at org.eclipse.smarthome.io.rest.core.internal.thing.ThingResource.updateConfiguration(ThingResource.java:415) [128:org.eclipse.smarthome.io.rest.core:0.10.0.201712141004]
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:498) ~[?:?]
        at org.glassfish.jersey.server.model.internal.ResourceMethodInvocationHandlerFactory$1.invoke(ResourceMethodInvocationHandlerFactory.java:81) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher$1.run(AbstractJavaResourceMethodDispatcher.java:144) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.invoke(AbstractJavaResourceMethodDispatcher.java:161) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.model.internal.JavaResourceMethodDispatcherProvider$ResponseOutInvoker.doDispatch(JavaResourceMethodDispatcherProvider.java:160) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.dispatch(AbstractJavaResourceMethodDispatcher.java:99) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.model.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:389) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:347) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:102) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.ServerRuntime$2.run(ServerRuntime.java:326) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.internal.Errors$1.call(Errors.java:271) [179:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors$1.call(Errors.java:267) [179:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:315) [179:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:297) [179:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:267) [179:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:317) [179:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.server.ServerRuntime.process(ServerRuntime.java:305) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.ApplicationHandler.handle(ApplicationHandler.java:1154) [180:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.servlet.WebComponent.serviceImpl(WebComponent.java:473) [177:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.WebComponent.service(WebComponent.java:427) [177:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:388) [177:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:341) [177:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:228) [177:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at com.eclipsesource.jaxrs.publisher.internal.ServletContainerBridge.service(ServletContainerBridge.java:76) [17:com.eclipsesource.jaxrs.publisher:5.3.1.201602281253]
        at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:848) [90:org.eclipse.jetty.servlet:9.3.22.v20171030]
        at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:584) [90:org.eclipse.jetty.servlet:9.3.22.v20171030]
        at org.ops4j.pax.web.service.jetty.internal.HttpServiceServletHandler.doHandle(HttpServiceServletHandler.java:71) [193:org.ops4j.pax.web.pax-web-jetty:6.0.7]
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:548) [87:org.eclipse.jetty.security:9.3.22.v20171030]
        at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:226) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1180) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.ops4j.pax.web.service.jetty.internal.HttpServiceContext.doHandle(HttpServiceContext.java:284) [193:org.ops4j.pax.web.pax-web-jetty:6.0.7]
        at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:512) [90:org.eclipse.jetty.servlet:9.3.22.v20171030]
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1112) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.ops4j.pax.web.service.jetty.internal.JettyServerHandlerCollection.handle(JettyServerHandlerCollection.java:80) [193:org.ops4j.pax.web.pax-web-jetty:6.0.7]
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:134) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.server.Server.handle(Server.java:534) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:333) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:251) [89:org.eclipse.jetty.server:9.3.22.v20171030]
        at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:283) [81:org.eclipse.jetty.io:9.3.22.v20171030]
[... cut ...]
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:671) [92:org.eclipse.jetty.util:9.3.22.v20171030]
        at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:589) [92:org.eclipse.jetty.util:9.3.22.v20171030]
        at java.lang.Thread.run(Thread.java:748) [?:?]
```

I just updated a few things in the database for these devices (Beaming, Security, and Max Baud), but doubt this will correct this issue when they make it to the binding. I have also deleted the Things to make sure I was on the latest Thing types from the binding.

cdjackson, 2018-01-13:
This is a strange one - it's not an error in the binding - at least the stack trace is not linked to the binding at all.

Can you capture what is being sent from HABmin using the debug console in your browser?

openhab-5iver, 2018-01-13:
This is for a working device:
[image]

This is for a device that gets the exception:
[image]

So it looks like the same thing is being sent. Using the REST API directly, I get the same response.

cdjackson, 2018-01-13:
Looks fine.

Is the status of the non-working devices online?

openhab-5iver, 2018-01-13:
Yes, they are all online and functioning happily. No other issues with them.

cdjackson, 2018-01-13:
Both these devices have no configuration - I suspect it's related.

cdjackson, 2018-01-13:
Found the problem - it's in ESH...

cdjackson, 2018-01-13:
https://github.com/eclipse/smarthome/pull/4914 opened in ESH.
Closing as this is not an issue with the binding.

closed 2018-01-13 by cdjackson